intel-aero/meta-intel-aero #296: How to connect intel aero fc with another telem

Huibean, 2017-12-13:
Currently I need to connect aero fc with another telem port, because it was using another protocol not mavlink. I check the docs that telem port is connect to fc's uart5
[image]
[image]
i set this port protocol to Mavlink by QgroundControl and connect to this port by usb to part with my mac, but i found that is not mavlink output. So, how can i connect intel aero fc with another telem? I need to keep tty/S1 connection with mavlink

sugnanprabhu, 2017-12-13:
@Huibean can you share the output of get_aero_version.py

sugnanprabhu, 2017-12-13:
Using v1.6 release, I'm able to connect to QGC using a ttl .

Following is the current order of telemetry pins.
    1 - VCC +5V
    2 - Telemetry TX
    3 - Telemetry RX
    4 - I2C CLK
    5 - I2C SDA
    6 - GND

sugnanprabhu, 2017-12-13:
Using v1.6 release, I'm able to connect to QGC using UART to ttl converter.

Following is the current order of telemetry pins.
    1 - VCC +5V
    2 - Telemetry TX
    3 - Telemetry RX
    4 - I2C CLK
    5 - I2C SDA
    6 - GND

Huibean, 2017-12-13:
@sugnanprabhu It's wired since I change os to ubuntu 16, fc version are unknow, but it did flash adrupilot firmware successfully
`huibean@huibean-CherryTrail:/etc/aerofc/ardupilot$ sudo aero-get-version.py

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.3 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown

huibean@huibean-CherryTrail:/etc/aerofc/ardupilot$ uname -a
Linux huibean-CherryTrail 4.4.76-aero-1.2 #1 SMP PREEMPT Thu Nov 9 21:20:17 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

sugnanprabhu, 2017-12-14:
You can try copying the script from the following location.

https://github.com/intel-aero/meta-intel-aero-base/blob/master/recipes-support/scripts/files/aerofc-force-bootloader-pin.py

Huibean, 2017-12-14:
@sugnanprabhu I have used the way you mentioned above, it works. But after flashing fc successfully, I still get no mavlink msg output from the telem port, it is my aero version wrong?I did follow the newest [Ubuntu* installation](https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation) from wiki
```shell
huibean@huibean-CherryTrail:~$ sudo aero-get-version.py
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.3 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown
```
here is my config in fc
[image]

Huibean, 2017-12-14:
I have used the same way to test with pixhawk1, it works fine

sugnanprabhu, 2017-12-14:
@Huibean Getting AeroFC firmware has some know issue on Ubuntu, which might be fixed in the next release.

I tried telem2 on both on Ubuntu and Yocto.
On Ubuntu, I also tried with ardupilot firmware and I do not see any issue.
It works on both QGC and Mission planner.

I do not see any issue with the BIOS version and the FPGA version that you are using, try checking the FC firmware version you have loaded in QGC.
[image]

Huibean, 2017-12-14:
@sugnanprabhu I am using apm copter 3.5.3
[image]

sugnanprabhu, 2017-12-14:
I'm using same version and I do not see any issue.

I would suggest try flashing the px4 firmware
reset all the parameters in QGC and reflash the px4 firmware to recover the default parameters for the px4 aeroFC.

If it still doesn't work try connecting to nsh over the mavlink_shell and check mavlink status, this should show if the instance is running for the telem2 port.

nsh> mavlink status

instance #0:
	mavlink chan: #0
	no telem status.
	flow control:	ON
	rates:
	tx: 0.886 kB/s
	txerr: 0.082 kB/s
	rx: 0.000 kB/s
	rate mult: 0.644
	accepting commands: YES
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS3 @57600)

instance #1:
	GCS heartbeat:	658429 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control:	ON
	rates:
	tx: 62.823 kB/s
	txerr: 0.000 kB/s
	rx: 0.038 kB/s
	rate mult: 1.000
	ULog rate: 50.0% of max 70.0%
	accepting commands: YES
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS1 @921600)

Huibean, 2017-12-14:
@sugnanprabhu thanks ! it works now, also with ardupilot. By the way I made some mistake before, this telem is mapping with serial5 and it suppose not support telem radio
[image]

closed 2017-12-15 by sugnanprabhu

wangxy524, 2017-12-19:
@sugnanprabhu excuse me? After I copied the script, I still got the "command not found" error