intel-aero/meta-intel-aero #299: Spektrum radio channel binding are wrong in ardupliot

veshnu, 2017-12-26:
All the channels are mapped right with px4 but not with ardupilot. Only the throttle is right. FLAP switch is mapped to YAW channel. How do I resolve this issue?

lucasdemarchi, 2017-12-26:
What do you mean they are not right? I was flying with them as is... I was using QGC to calibrate that  maybe inverts some channels by its own, but the mapping itself should be right. Please open an issue on the ardupilot repository so we can figure it out there.

closed 2017-12-26 by lucasdemarchi

magicrub, 2017-12-27:
@veshnu Here's a link to the ArduPilot issues area: https://github.com/ArduPilot/ardupilot/issues

However, as @lucasdemarchi said, I suspect it's not actually a flight controller problem. Are you sure you haven't accidentally remapped them yourself? The default config should work fine