intel-aero/meta-intel-aero #300: repo signature problem?

guermonprez, 2017-12-27:
I would suspect a signature problem, or a problem with the hosting cluster.

installing on a clean system, the download is reporting errors:

paul@corsica:~$ sudo apt-get -y install aero-system
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  aero-ardupilot aero-bios aero-camera-streaming-daemon aero-fpga aero-init aero-mavlink-router aero-optical-flow aero-px4 aero-spi-xfer aero-systemd aero-utils firmware-atomisp i2c-tools
  libgstrtspserver-1.0-0 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-imgproc2.4v5 libopencv-video2.4v5 libtbb2 linux-headers-4.4.76-aero-1.2 linux-image-4.4.76-aero-1.2
  python-serial read-edid
Suggested packages:
  libi2c-dev python-smbus linux-firmware-image-4.4.76-aero-1.2 python-wxgtk3.0 | python-wxgtk
The following NEW packages will be installed:
  aero-ardupilot aero-bios aero-camera-streaming-daemon aero-fpga aero-init aero-mavlink-router aero-optical-flow aero-px4 aero-spi-xfer aero-system aero-systemd aero-utils firmware-atomisp i2c-tools
  libgstrtspserver-1.0-0 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-imgproc2.4v5 libopencv-video2.4v5 libtbb2 linux-headers-4.4.76-aero-1.2 linux-image-4.4.76-aero-1.2
  python-serial read-edid
0 upgraded, 25 newly installed, 0 to remove and 5 not upgraded.
Need to get 24.5 MB/26.7 MB of archives.
After this operation, 106 MB of additional disk space will be used.
Get:1 https://download.01.org/aero/deb xenial/main amd64 linux-image-4.4.76-aero-1.2 amd64 4.4.76-aero-1.2-1 [11.5 MB]
Err:1 https://download.01.org/aero/deb xenial/main amd64 linux-image-4.4.76-aero-1.2 amd64 4.4.76-aero-1.2-1
  Writing more data than expected (11508870 > 11505814)
Get:2 https://download.01.org/aero/deb xenial/main amd64 linux-headers-4.4.76-aero-1.2 amd64 4.4.76-aero-1.2-1 [6,988 kB]
Err:2 https://download.01.org/aero/deb xenial/main amd64 linux-headers-4.4.76-aero-1.2 amd64 4.4.76-aero-1.2-1
  Hash Sum mismatch
Get:3 https://download.01.org/aero/deb xenial/main amd64 aero-fpga amd64 0.1-3 [232 kB]
Err:3 https://download.01.org/aero/deb xenial/main amd64 aero-fpga amd64 0.1-3
  Writing more data than expected (232966 > 231920)
Get:4 https://download.01.org/aero/deb xenial/main amd64 aero-px4 amd64 0.1-3 [580 kB]
Err:4 https://download.01.org/aero/deb xenial/main amd64 aero-px4 amd64 0.1-3
  Writing more data than expected (580724 > 579754)
Get:5 https://download.01.org/aero/deb xenial/main amd64 aero-ardupilot amd64 0.1-3 [593 kB]
Err:5 https://download.01.org/aero/deb xenial/main amd64 aero-ardupilot amd64 0.1-3
  Writing more data than expected (594290 > 593346)
Get:6 https://download.01.org/aero/deb xenial/main amd64 aero-spi-xfer amd64 0.1-4 [4,122 B]
Err:6 https://download.01.org/aero/deb xenial/main amd64 aero-spi-xfer amd64 0.1-4
  Writing more data than expected (5058 > 4122)
Get:7 https://download.01.org/aero/deb xenial/main amd64 aero-bios amd64 0.1-4 [3,430 kB]
Err:7 https://download.01.org/aero/deb xenial/main amd64 aero-bios amd64 0.1-4
  Writing more data than expected (3431066 > 3430204)
Get:8 https://download.01.org/aero/deb xenial/main amd64 firmware-atomisp amd64 0.1-3 [1,144 kB]
Err:8 https://download.01.org/aero/deb xenial/main amd64 firmware-atomisp amd64 0.1-3
  Writing more data than expected (1145020 > 1144164)
Fetched 6,984 kB in 4s (1,537 kB/s)         
E: Failed to fetch https://download.01.org/aero/deb/pool/main/l/linux-4.4.76-aero-1.2/linux-image-4.4.76-aero-1.2_4.4.76-aero-1.2-1_amd64.deb  Writing more data than expected (11508870 > 11505814)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/l/linux-4.4.76-aero-1.2/linux-headers-4.4.76-aero-1.2_4.4.76-aero-1.2-1_amd64.deb  Hash Sum mismatch

E: Failed to fetch https://download.01.org/aero/deb/pool/main/a/aero-fpga/aero-fpga_0.1-3_amd64.deb  Writing more data than expected (232966 > 231920)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/a/aero-px4/aero-px4_0.1-3_amd64.deb  Writing more data than expected (580724 > 579754)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/a/aero-ardupilot/aero-ardupilot_0.1-3_amd64.deb  Writing more data than expected (594290 > 593346)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/a/aero-spi-xfer/aero-spi-xfer_0.1-4_amd64.deb  Writing more data than expected (5058 > 4122)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/a/aero-bios/aero-bios_0.1-4_amd64.deb  Writing more data than expected (3431066 > 3430204)

E: Failed to fetch https://download.01.org/aero/deb/pool/main/f/firmware-atomisp/firmware-atomisp_0.1-3_amd64.deb  Writing more data than expected (1145020 > 1144164)

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

avinash-palleti, 2017-12-28:
I tried just now, it is working. Can you please try again.

avinash-palleti, 2018-01-11:
No issue is observed with latest release.

closed 2018-01-11 by avinash-palleti