intel-aero/meta-intel-aero #306: Compilation error while building the aero image with gcc 7

Trass3r, 2018-01-15:
Basically this patch is currently necessary:
https://lists.01.org/pipermail/edk2-devel/2017-February/007329.html

```sh
repo init -u https://github.com/intel-aero/intel-aero-manifest.git -m v1.6.1.xml -b master
repo sync -j4

export TEMPLATECONF=../meta-intel-aero/conf/
source poky/oe-init-build-env
 
bitbake intel-aero-image
sed -ri "s/mStringFileName == '/mStringFileName == NULL || *mStringFileName == '/" tmp/work/x86_64-linux/ovmf-native/git-r0/git/BaseTools/Source/C/VfrCompile/VfrUtilityLib.cpp
 
bitbake intel-aero-image
```

sugnanprabhu, 2018-01-15:
@Trass3r Right way to apply additional patches would be to add .bbappend

Following is one of the example.
https://github.com/intel-aero/meta-intel-aero/blob/master/recipes-devtools/rpm/rpm_%25.bbappend

Trass3r, 2018-01-15:
It's just a quick hack. The omvf repository should be updated instead.

avinash-palleti, 2018-01-15:
> @Trass3r Right way to apply additional patches would be to add .bbappend

 I think there is little confusion here, patch what @Trass3r suggested is to be applied on host machine where the build is happening. bbappend is not applicable here.

@Trass3r We have still not moved to gcc 7 in our host machine, that is the reason we have not encountered this issue. We will update the documentation with the proper fix to overcome this issue . Thanks for suggesting.

avinash-palleti, 2018-01-16:
@Trass3r We have still not moved to gcc-7 that's why may be we didn't encounter this issue.
I see the patch which  you gave is outdated, in ovmf repository `mStringFileName` is already compared with `NULL`.
https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/VfrCompile/VfrUtilityLib.cpp#L3509

Trass3r, 2018-01-16:
Indeed. The other existing patches need to be updated.

anselmolsm, 2018-01-16:
> It's just a quick hack. The omvf repository should be updated instead.

omvf recipes are part of Yocto Project's poky. omvf has been updated in [pyro branch already](http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?h=pyro&id=333b79aea38385dfb52fcd24e121bca7d27e68dd).

The manifest file for 1.6.1 can't be modified (thus it needs gcc 6).
For build with gcc 7 you can use the devel manifest file (default.xml) instead as it tracks the HEAD of the poky pyro branch.
Alternatively, you can just update your local poky clone to get the patch linked above.
Thanks

closed 2018-01-16 by anselmolsm