intel-aero/meta-intel-aero #309: Not able to connect with QGroundControl

mandelyoni, 2018-02-19:
Hi, 

I'm not able to connect with QGroundControl, after installing Ubuntu,  according to  90 (References) OS user Installation · intel-aero/meta-intel-aero Wiki · GitHub 
I receive the attached response:
"vehicle did not respond to request for parameters. This will cause qgroundcontrol to be unable to display its full interface"
[image]

lmicko, 2018-02-19:
Hi, I've faced the same issue. Didn't have much time to figure out what's wrong, but using previous version of QGC worked well.

mandelyoni, 2018-02-19:
it worked
version 3.2.7 of QGC is good

thank you

anitha-suresh, 2018-02-20:
Closing this issue, as I see it works for Yoni now.

closed 2018-02-20 by anitha-suresh