intel-aero/meta-intel-aero #311: Pixhawk2

Thanos254, 2018-02-20:
Can I connect PIxhawk2 FC with the Intel Aero?

anitha-suresh, 2018-02-20:
@Thanos254, yes!

@sugnanprabhu , can you please provide your inputs

sugnanprabhu, 2018-02-20:
Please check the following documentation which is for pixhawk which is applicable for pixhawk2 as well.
https://github.com/intel-aero/meta-intel-aero/wiki/93-(References)-Pixhawk-FC

JustFineD, 2018-02-20:
Hi, I was able to connect pixhawk1 to the aero using UART. 
Unfortunately when I tried to connect pixhawk2 (cube) - I wasn't able to make a connection to the aero

sugnanprabhu, 2018-02-21:
@JustFineD what pins are you connecting from pixhawk2 to compute board?
You can try connecting the TELEM pin, make sure to use the correct baud rate and telmetry pin is configured to get the mavlink messages.

closed 2018-06-05 by zehortigoza