intel-aero/meta-intel-aero #312: Get image from MIPI camera OV7251 nad OV8858 local not stream

Thanos254, 2018-02-20:
Dear All, we are looking to get the image (stream ) from two MIPI cameras OV7251 (global shutter VGA) OV8858 (rolling shutter HD ) on Intel Aero board on local monitor that is connected with HDMI cable to the Aero board with gstreamer. Please tell us how can we do this?  We are aware of 06 Cameras and Video · intel-aero/meta-intel-aero Wiki · GitHub  and we don't want to use streaming over WiFI. We want to use command like this: gst-launch-1.0 v4l2src /dev/videox (video x is eg video1 ) ! autovideoconvert ! autovideosink  OR gst-launch-1.0 v4l2src /dev/videox ! format ! xvimagesink How can we do this ?? We are using: BIOS_VERSION = Aero-01.00.13OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.1 (pyro)"AIRMAP_VERSION = 1.8FPGA_VERSION = 0xc2AeroFC firmware version = 1.6.5 We noticed that there are some error in dmesg with OV7251. [    3.574667] ov7251 i2c-INT35AA:00: gmin: initializing atomisp module subdev data.PMIC ID 1[    3.574722] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_CamClk[    3.574879] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_ClkSrc[    3.574930] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_CsiPort[    3.575070] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_CsiLanes[    3.575378] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo1_1p8v[    3.575543] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo1_sel_reg[    3.575682] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo1_ctrl_shift[    3.575821] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo2_1p8v[    3.575961] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo2_sel_reg[    3.576100] acpi INT35AA:00: Failed to find gmin variable INT35AA:00_eldo2_ctrl_shift[    3.609534] ov7251 i2c-INT35AA:00: camera pdata: port: 0 lanes: 1 order: 00000002[    3.613538] ov7251 i2c-INT35AA:00: sensor_revision = 0x0[    3.613542] ov7251 i2c-INT35AA:00: detect ov7251 success[    3.620090] input: ov7251 as /devices/virtual/input/input4[    3.620442] ov7251 i2c-INT35AA:00: register atomisp i2c module type 1[    3.626493]  tal test probe called[    3.632176] ov7251 i2c-INT35AA:01: gmin: initializing atomisp module subdev data.PMIC ID 1[    3.632436] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_CamClk[    3.632607] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_ClkSrc[    3.632659] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_CsiPort[    3.632805] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_CsiLanes[    3.633116] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo1_1p8v[    3.633258] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo1_sel_reg[    3.635467] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo1_ctrl_shift[    3.635612] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo2_1p8v[    3.635754] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo2_sel_reg[    3.635895] acpi INT35AA:01: Failed to find gmin variable INT35AA:01_eldo2_ctrl_shift[    3.668474] ov7251 i2c-INT35AA:01: camera pdata: port: 1 lanes: 1 order: 00000002[    3.669134] ov7251 i2c-INT35AA:01: read from offset 0x300a error -121[    3.676402] ov7251 i2c-INT35AA:01: sensor_id_high = 0x0[    3.682287] ov7251 i2c-INT35AA:01: ov7251_detect err s_config.[    3.691652] ov7251 i2c-INT35AA:01: sensor power-gating failed Please can you look at this? Also please can you share the correct dmesg from your aero platform?
--

lbegani, 2018-02-22:
It is not possible to take images from the two cameras by running gst-pipeline in command line.
Read here how you can do it - https://github.com/intel-aero/meta-intel-aero/wiki/06-Cameras-and-Video#take-a-photo-with-the-black-and-white-camera

Thanos254, 2018-02-23:
thank you for the replay, we will try to capture the image from OV7251 as you suggested with the sample app.

Regarding the OV8858, ok there are some issues with underexposed picture.

But can you tell me where is the sample app for the HD OV8858 camera ?

Or can we get the stream ( doesn't mater if underexposed) with gstreamer from the OV8858?

Which device corresponds to the OV8858 camera?

zehortigoza, 2018-03-29:
Example of how get frames from ov8858 https://github.com/intel-aero/sample-apps/tree/master/hd-camera but the quality is not good at all, the driver needs some changes.

closed 2018-03-29 by zehortigoza