intel-aero/meta-intel-aero #314: AIRMAP_VERSION = unknown

Thanos254, 2018-02-20:
I installed ubuntu on intel aero RTF and when I run the aero-get-version.py command  it's saying the AIRMAP_VERSION = unknown

I know Iit is related to Linux kernel. I Choose Linux 4.4.xx to log in to ubuntu.

Here are reference links:-

https://communities.intel.com/external-link.jspa?url=https%3A%2F%2Fgithub.com%2Fintel-aero%2Fmeta-intel-aero%2Fissues%2F291

https://communities.intel.com/external-link.jspa?url=https%3A%2F%2Fgithub.com%2Fintel-aero%2Fpackages%2Fissues%2F37

https://communities.intel.com/thread/122646

I am still getting AIRMAP_version unknown.

avinash-palleti, 2018-02-20:
Hi @Thanos254 ,

Hope you have followed full steps as per this [wiki](https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation) 
Please share the complete output of below commands,
`aero-get-version.py`
`dpkg -s aero-system`
`uname -r`

szebedy, 2018-06-02:
Hi @avinash-palleti ,

I have followed the instructions as per wiki, and I'm getting the same output.

**$ get_aero_version.py**

/sys/firmware/dmi/tables/smbios_entry_point: Permission denied
/dev/mem: Permission denied
BIOS_VERSION = unknown
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = unknown
AeroFC firmware version = 1.6.5

**$ sudo !!**
sudo get_aero_version.py

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.6.5

**$  dpkg -s aero-system**
Package: aero-system
Status: install ok installed
Priority: important
Section: misc
Installed-Size: 9
Maintainer: Daniel E Ruano <daniel.e.ruano@intel.com>
Architecture: all
Multi-Arch: foreign
Version: 1.0.10
Depends: linux-image-4.4.76-aero-1.2, linux-headers-4.4.76-aero-1.2, aero-mavlink-router, aero-fpga, aero-px4, aero-ardupilot, aero-utils, aero-init, aero-bios, openssh-server, i2c-tools, firmware-atomisp, aero-systemd, aero-camera-streaming-daemon, aero-optical-flow
Description: Collection of Intel Aero packages.
 This collection of Intel Aero packages includes all necessary programs and
 utilities to use the Intel Aero RTF drone on Ubuntu.
Homepage: https://github.com/intel-aero/

**$ uname -r**
4.4.76-aero-1.2