intel-aero/meta-intel-aero #316: ROS installation problem

mandelyoni, 2018-02-21:
Hi,

while installing ROS according to https://github.com/intel-aero/meta-intel-aero/wiki/05-Autonomous-drone-programming-with-ROS,    we receive the error in the pic attached.

Any idea what to do?

Thank you

[image]
?

Thank you,  Yoni.

shakthi-prashanth-m, 2018-02-21:
Hi Yoni,
If you experience issues connecting to the keyserver, you can try substituting hkp://pgp.mit.edu:80 or hkp://keyserver.ubuntu.com:80 in the previous command.

as suggested in ROS-Kinetic Ubuntu installation guide: http://wiki.ros.org/kinetic/Installation/Ubuntu

mandelyoni, 2018-02-26:
Thanks, I wrote this:

sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116

shakthi-prashanth-m, 2018-02-27:
OK! @mandelyoni were you able to install successfully ?

mandelyoni, 2018-02-27:
Yes, thank you

shakthi-prashanth-m, 2018-02-27:
Closing as issue resolved.

closed 2018-02-27 by shakthi-prashanth-m

mymultiverse, 2018-10-05:
I am getting following message.

```
gpg: requesting key B01FA116 from hkp server pgp.mit.edu
gpg: key B01FA116: "ROS Builder <rosbuild@ros.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

```

```
gpg: requesting key B01FA116 from hkp server keyserver.ubuntu.com
gpg: key B01FA116: "ROS Builder <rosbuild@ros.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```