intel-aero/meta-intel-aero #317: memory leak in the Intel Aero

shai91521, 2018-02-21:
Hi, 

**I followed your instructions on** 
https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#choice-of-oses
**up to (and including)** 
https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#intel-realsense-sdk 

and then I noticed there is a memory leakage - please see attached pictures at consecutive times (look at the watch) :

[image]
[image]

What do you think?

Thank you

lucasdemarchi, 2018-02-21:
This is a picture showing the storage used, not ram.

closed 2018-02-21 by lucasdemarchi

mandelyoni, 2018-02-21:
I know
It still doesn't solve the problem  :)

anselmolsm, 2018-02-21:
There is a problem in the problem statement.
Memory leak is related to RAM, not storage.
The only conclusion from the pictures is that the system is using +100MB of **storage** after 57 minutes.