intel-aero/meta-intel-aero #318: No output from hdmi

Thanos254, 2018-02-23:
Hi, I used intel drone very well.I installed ubuntu successfully, and I used it well. However, it suddenly did not output any signal from hdmi.After turning on the intel drone, the computer seems like to be powered on, but I cannot see any output from the hdmi.Also, to escape from the sleep mode or something, I pushed the power button after turning on intel drone.But it does not work.(Pushing the power button for a few second to power off does not work neither.) To show LED light patter, I give two links.

https://www.youtube.com/watch?v=2TBXhJqpeJw&feature=youtu.be

https://www.youtube.com/watch?v=3WGj1xQL1zg&feature=youtu.be

zehortigoza, 2018-03-20:
Can you SSH into the board?

closed 2018-06-05 by zehortigoza

PhilippeDoucette, 2018-07-13:
My RTF did the same thing.  After flashing the update, I re calibrated it using Q Ground Control.  After that, the HDMI does not work and it would not turn off with the power button.  I unplugged it and now it will boot, I can connect with wifi to Q Ground Control, and I can ssh into it. But still no HDMI.   How did you solve this?

PhilippeDoucette, 2018-07-13:
It is part of the Intel RTF drone.  Which is powered by by a power supply, and has been working fine, through the full flashing process documented here: 
https://www.youtube.com/watch?v=e9MLnRbMDHo&list=PLTQSXsG86pGfyZm5ac6-ZtQsEniUJIE9o&index=7
and here
https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-Setup

The procedure went perfectly, but when complete, the HDMI no longer works.