intel-aero/meta-intel-aero #320: PX4Flow connectivity issue

trgiman, 2018-03-01:
Hi all,

We have connected px4flow module to telemetry port of Intel Aero. Previously we connected in the same way on the same port Garmin Lidar. We are absolutely sure, that SDA and SCL and VDD and GND cables are correct. px4flow module works well when connected trough USB port to QGControl 3.3.0 and we uploaded firmware and performed calibration....

Current configuration of Intel RTF is:

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.1 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.7.3

Following are the issues:

1th try: We were following guide for connecting px4flow to Intel RTF according Intel Instructables. But after uploading the custom PX4 firmware file aerofc-v1_px4flow.px to Aero FC QGround 3.3.0 is not able to connect, also mavlink_shell.py command does not allows us connect to flight controller nsh shell. For AeroFC fmw version 1.6.5 works everything well.

2nd try: We compiled our own PX4 firmware for AeroFC and added drivers/px4flow into the nuttx_aerofc-v1_default.cmake...but results are.....when we apply command px4flow start. It just says "scanning I2C buses for device..than it stops" When we run px4flow info. It says driver not running. 

Could you please provide us with correct configuration for point  1,2? At the end of the day we would like to use both Garmin Lidar and px4flow at once to improve quality of flight without GPS.

> Current answers from Jose:

Telemetry pins
5V
UART TX
UART RX
I2C CLK
I2C SDA
GND

I would not recomend to use the firmware from the Instructables as it is really old and uses a different baudrate(that is why QGroundControl can not connect to it)
There is the px4flow driver in regular Aero PX4 firmware?! that is odd.
Make sure you are using the Aero-rtf firmware that comes with the Aero image.
Just to make sure, do not keep the USB cable connected to PX4Flow when connecting it to Aero.

Other option would be connect the PX4Flow using UART, there is a I2C port and UART port in it. You can connect the UART port to the UART RX and UART TX of the telemetry port but you need to change the baudrate of that port in PX4, right now it is runing at 57600 and PX4Flow runs at 115200.

Thank you,
Martin

trgiman, 2018-03-01:
Hi @zehortigoza. Thanks a lot for answers. I am new to PX4 settings. I did find following file. But I am a little bit lost /Firmware/ROMFS/px4fmu_common/init.d/rcS ...mapping of TELEM is not obvious for AERO FC for me. Could you please advice? 

Eventually if possible I would like to still try to connect PX4flow trough i2c versus UART.

sugnanprabhu, 2018-03-02:
@trgiman px4flow px4flow driver is not started by default for aero.

https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L682

Try adding the following.

```
+       if ver hwcmp AEROFC_V1
+       then
+               px4flow start &
+       fi

```

sugnanprabhu, 2018-03-02:
@trgiman Currently px4flow sensor driver is not look for the slave address on the i2c available on the telemetry port. Please try the following change along with the above changes.

https://github.com/PX4/Firmware/pull/8994

sugnanprabhu, 2018-03-02:
Also as you have mentioned earlier, you need to enable px4flow driver in the following file.

trgiman, 2018-03-02:
Hi @sugnanprabhu it was helpful. Now when I run px4flow start from nsh shell. I have positive answer.Perfect. Thanks....but there are issues....

nsh> px4flow start
WARN  [px4flow] scanning I2C buses for device..
PX4FLOW on I2C bus 2 at 0x42 (bus: 100 KHz, max: 400 KHz)

**But it works only when px4flow is connected to USB power source and Telem port for i2c communication.** I already checked wires an checked, that Voltage between pin4 (GND) and pin1 (VDD) of i2c Pix4Flow port is cca 5.3V which should be ok. In the same telem port Garmin Lidar was working beeing powered directly from TELEM port.

PX4flow has got the most recent stable Firmware. Infofrom QGControl
Found device: PX4 Flow
Connected to bootloader:
Version:3
Board ID:6
Flash size: 1032192

But it seems the bootloader param is empty?

I tried to power Intel Aero RTF from bench powersupply 16v 5.6A and also from battery 4s (14.8V) and 4000maH. There is not change in behavioure

trgiman, 2018-03-03:
@sugnanprabhu it turned out to be faulty PX4flow. **Otherwise your recommendations helped to solve the issue**.Thanks a lot. Could you please advice where in file /Firmware/ROMFS/px4fmu_common/init.d/rcS we could change baud speed for Intel Aero Telemetry USART part?

Generally is there some documentation that shows how compass port, telemetry port,gps,radio etc. relate to PX4 firmware /Firmware/ROMFS/px4fmu_common/init.d/rcS settings? Also is it correct, that copass port also supports i2c communication? Simply where we can find documentation for PX4 firmware to Aero RTF ports mapping?

@zehortigoza original recommendation:

Other option would be connect the PX4Flow using UART, there is a I2C port and UART port in it. You can connect the UART port to the UART RX and UART TX of the telemetry port but you need to change the baudrate of that port in PX4, right now it is runing at 57600 and PX4Flow runs at 115200.

sugnanprabhu, 2018-03-05:
@trgiman I think it is ttyS2.
I would suggest try to list the mavlink instances that are running and find the device node that is used by the mavlink instance for TELETRY 2.

There is not specific documentation about the mapping of pins from PX4 firmware to Aero subPCB, as the PX4 is connected via Altera MAX10 FPGA.

Please check the following documentation if it is useful.
https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/intel-aero-compute-board-guide.pdf

trgiman, 2018-03-05:
@sugnanprabhu I connected Intel Aero RTF TELEM port using FTDI cable to RX TX pin. Than I was able to connect to nsh shell. Output of mavlink status is:

nsh> mavlink status

instance #0:
	GCS heartbeat:	226530 us ago
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control:	ON
	rates:
	tx: 0.150 kB/s
	txerr: 0.000 kB/s
	rx: 0.021 kB/s
	rate mult: 0.671
	accepting commands: YES, FTP enabled: NO
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS3 @57600)

instance #1:
	GCS heartbeat:	1375919515 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control:	ON
	rates:
	tx: 42.093 kB/s
	txerr: 0.000 kB/s
	rx: 0.000 kB/s
	rate mult: 1.000
	ULog rate: 50.0% of max 70.0%
	accepting commands: YES, FTP enabled: YES
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS1 @921600)

So /dev/ttyS3 device in Nuttx is rx tx pin on the TELEM port. Since /dev/ttyS1 is used when connecting wireless or USB3 on AeroBoard.Now question is where to disable Mavlink for /dev/ttyS3, change baudrate and connect PX4Flow trough TELEM RX,TX. Will PX4Flow driver work trough TX RX? 

Should I change it somewhere here /Firmware/ROMFS/px4fmu_common/init.d/rcS? Thank you. By the way it seem I can manually stop all mavlink instances but not 1 single mavlink instance as mavlink stop is no longer valid command.

closed 2018-03-17 by trgiman

Tim28rc, 2018-07-23:
I'm also having trouble with connecting the px4flow to the telem port. When I try to start the px4flow driver through nsh in the qgc it just says "scanning i2c buses for device" but never connects. Initially when I connected the px4flow by splitting the compass i2c port, the px4flow driver was working fine, but when I switched, now it seems it can't find it. That said I am splitting the telem port between the px4flow and a leddarOne. Above it says to switch the baudrate, but I can't find where to actually do that in the px4 source code. Also, would switching the baudrate disable the use of the leddar one on the same port?
Thanks