intel-aero/meta-intel-aero #325: pxflow not working

mandelyoni, 2018-03-12:
Hi,

I'm working according  to :

Intel Aero Drone - Altitude and Position Hold Using PX4Flow: 11 Steps (with Pictures) 

but I have no connection with the QGC after all the installations.

Any idea?

Thanks,  Yoni.

trgiman, 2018-03-12:
hi @mandelyoni all solutions you need to solve it are here https://github.com/intel-aero/meta-intel-aero/issues/320

mandelyoni, 2018-03-12:
Is it working on your drone?
It looks like the problem isn't solved

mandelyoni, 2018-03-17:
Hi, can someone please assist?

mandelyoni, 2018-03-17:
Hi,
Was someone able to operate the px4flow according to
http://www.instructables.com/id/Intel-Aero-Drone-Altitude-and-Position-Hold-Using-/ ?

What firmare did you use on the PX4 and the FPGA?
What QGC version?
any other modifications?

Thank you

trgiman, 2018-03-17:
Hi @mandelyoni  yes it works. Tutorial you mentioned is old in regards of software setup. Follow https://github.com/intel-aero/meta-intel-aero/issues/320

mandelyoni, 2018-03-17:
Great, thank you

mandelyoni, 2018-03-17:
Hi Trgiman,

The instructions at #320  are not so clear to us.
Can you please summarize your solution?

Thank you very much,  Yoni,

closed 2018-03-17 by mandelyoni

sluthra1, 2018-04-12:
Hello, I am a little confused on what was done in #320, can someone clarify the steps they took to get px4flow working?