intel-aero/meta-intel-aero #331: GadgetFS disables keyboard/mouse

jiajun78, 2018-03-21:
Hi,

I have encountered some issues using gadgetfs on the Intel Aero Compute Board.

My setup is as follows:
OS: ubuntu 16.04
Kernel: 4.4.76-aero-1.2
I downloaded the kernel source from https://github.com/intel-aero/linux-kernel
Using the config from https://github.com/intel-aero/meta-intel-aero-base/blob/master/recipes-kernel/linux/linux-yocto/defconfig

I configured the kernel to use gadgetfs instead of ethernet gadget (g_ether).

On boot, the mouse and keyboard will not work at the Ubuntu login screen.

This is rather odd.
Using a kernel with g_ether instead of gadgetfs will not encounter this problem.
This seems to be an issue with gadgetfs on the board.

Any help is appreciated.

zehortigoza, 2018-03-23:
No one have tried this and I'm not sure if the driver supports it:
https://github.com/intel-aero/linux-kernel/commit/9c7592a46553ca8bfcd02f204e6db8bd8380eca0
Maybe if you backport the upstream driver to Aero kernel(4.4) it might work.

jiajun78, 2018-05-04:
Managed to solve the issue using kernel 4.4.76-aero-1.2

It was due to a bug in the userspace gadget driver ..

Thanks

closed 2018-05-04 by jiajun78