intel-aero/meta-intel-aero #332: Using teraranger one with intel aero optical flow

mubaarik, 2018-03-21:
I have been trying to connect a [teraranger One](https://www.terabee.com/wp-content/uploads/files/TeraRanger_to_Pixhawk_Connection_Instructions.pdf) to the intel aero for the optical-flow vertical distance sensing. The set up is similar to the one [here](https://docs.px4.io/en/flight_controller/intel_aero.html) with the Lidar Lite3 using the compass port. I am using [this adaptor](http://www.teraranger.com/product/adapters-for-oneduo/) to connect the sensor to the flight controller and powering it from the battery. 

The sensor turns on as the LEDs show. I have probed the SDA and SCL lines using oscilloscope and I can see data on the lines. Of course I didn't decode the data to see what's happening if anything but yes.

However QGroundcoontrol doesn't show connection. Then I have used the px4 console running the following commands with corresponding outputs 
```
nsh> teraranger status

error: [teraranger] driver not running
```
```
nsh> teraranger start

ERROR [teraranger] driver start failed.
```
After that I am not sure how to proceed. Any suggestions?

Thanks

zehortigoza, 2018-03-23:
So are you trying to connect this range sensor to the USB port using this? http://www.teraranger.com/product/adapters-for-oneduo/
If so the USB is not connected to the flight controller, it is connected to Aero compute board, so you can write a linux application to read that port and send mavlink messages to the flight controller with the distance data.

Or you should connect it to the telemetry port like in the like you send but you will need to add support to PX4_I2C_BUS_EXPANSION1(telemetry I2C is this one) in teraranger driver, see the Lidar Lite3 driver. Or adding a I2C splitter to the external compass port it should work PX4_I2C_BUS_EXPANSION but I would recommend you keep just the external compass in that port for performance reasons.

mubaarik, 2018-03-23:
I am using the I2C adaptor, no USB. I currently have the teraranger connected to the compass port with the compass disconnected. Any more suggestions? Also could you put a pointer to the modified Lidar Lite drivers if possible? And an example code for sending mavlink messages to the flight controller if you know one? 

Thanks

zehortigoza, 2018-03-29:
I still don't get what is this I2C adapter, the compass port is I2C the sensor is I2C, why the adapter? 
Just look at the Lidar lite driver at PX4, the upstream version of it support it in both expansion ports.
About mavlink examples you can check this ones: https://github.com/intel/mavlink-router/tree/master/examples

mubaarik, 2018-04-04:
It works on the Telemetry port, but not on the compass port which I have no idea why. The distance readings are full of spikes though even after installing decoupling capacitors as suggested for the Lidar Lite.

The adaptor is just there to give the sensor the proper voltages. The sensor expects 5V for SCL and SDA, but the Intel Aero I2C SDA and SCL are 3.3V right? 

Thanks

zehortigoza, 2018-04-04:
I have no idea why it is working in telemetry port and not in compass it uses PX4_I2C_BUS_EXPANSION...

Most sensors are powered with 5v and the communication is 3.3V, did had give a try?
Just remember to share the GND between the adaptor and the AeroFC connector.

It have a CRC byte you can change the code to use that so you will eliminate all the spikes: https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/teraranger/teraranger.cpp#L643

closed 2018-06-05 by zehortigoza

AlexisTM, 2018-09-06:
@zehortigoza The Teraranger One sensor uses a proprietary pinout.
The I2C extension has to be powered from 12V + I2C connector; The laser is not pulling current from I2C power lines.
It can alternatively be connected to a USB extension which is USB powered and simulates a UART port.