intel-aero/meta-intel-aero #333: Cannot connect Pixhawk2 to Aero computer board via TTL Serial Port

orshefi, 2018-03-22:
Followed this example to connect pixhawk serial mavlink to aero, it worked with pixhawk 1

When I tried to connect it with pixhawk 2 I couldn't get it to work.

I followed this tutorial:

https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/apu-161110-pixhawk-flight-guide.pdf

Also I checked the pixhawk2 firmware and hardware when I connected it to TTL FTDI cable and connected to the virtual comport with Mavproxy, the serial connection worked and the communication was good.

I have also tried all the serial ports available on the Aero SBC

Also I probed the serial line with an ocilloscope and the signal seemed to be clear and steady, there were no mismatches in parity or stop bits as well.

lucasdemarchi, 2018-03-22:
what do you mean by "all the serial ports"? There's only one available

orshefi, 2018-03-22:
I was just making sure the fpga isn't accidentally configured to route
other serial through these pins, I worked with /dev/ttyS1

zehortigoza, 2018-03-23:
Did you stopped the mavlink-router?
Other thing to try is to revert the workarround to make the UART to work at 921600bps: 
https://github.com/intel-aero/meta-intel-aero/commit/7ba677e246a9c84723ae3d62d2fd5c94ec05784f
It will reduce the error rate in lower baudrates.

closed 2018-06-05 by zehortigoza