intel-aero/meta-intel-aero #336: using the bottom camera of aero drone in linux

galbarnahum, 2018-04-11:
I have an intel aero drone, with ubuntu 16.04 on it.
I have tried to get a video from the cameras, but i couldnt get video from the bottom and angeled camera.
im adding a link to the api of the cameras-and-video: 
[https://github.com/intel-aero/meta-intel-aero/wiki/06-Cameras-and-Video](url)
As written there, "On Yocto (v1.6 minimum), Intel Aero is proposing 5 RTSP video feeds", 
and "On Ubuntu, Intel Aero is proposing 3 RTSP video feeds".
is there a way to use the other two cameras on ubuntu?

lbegani, 2018-04-12:
@galbarnahum The number of RTSP streams and the URI to access them is same for both yocto and Ubuntu. I have updated the page now.

szebedy, 2018-05-27:
Hi @galbarnahum,

Were you able to make the bottom/front angled cameras work? For me there is no response from the RTSP feeds

AlexisTM, 2018-09-05:
@galbarnahum @szebedy I could not have it working, probably because of a conflict between the different apps: Optical flow and CSD (camera streaming deamon).

So I made my own node to stream it to ROS. https://github.com/AlexisTM/aero-optical-flow-ros

It is not the same purpose as it is meant to send complete images with lower frequency for debug and have the full speed stream for extra processing.

zehortigoza, 2018-09-06:
@AlexisTM I did a quick look in your repository and looks like it is still computing optical-flow as aero-optical-flow does, this is just a first step?
I guess if you want to send the image to ROS world you would use one of the optical-flow or visual odometry algorithms.
If we come up with something more cleaner I can add it to intel-aero repositories.

AlexisTM, 2018-09-06:
@zehortigoza This is, of course, the first step. It is not ready to be used in production. This is the `intel-aero/aero-optical-flow` app with the following modifications:
- Built with catkin 
- 640x480 image is read from the camera
- Image is cropped to 630x470 (because the frame is buggy on the last pixels on the right and bottom)
- Image is published to ROS 
- Camera Info services are used to allow recalibration and keeping parameters

That's the minimal node to make it usable from ROS while keeping the original functionality. 

I currently use with the `ar_track_alvar` library for tag recognition.

The ideal library would be done with nodelets (one process with multiple apps for zerocopy image transport layer)