intel-aero/meta-intel-aero #337: Mavlink-router service does not start

ivanalexo, 2018-04-12:
Hi all,

I've installed Ubuntu on my RTF drone and followed the steps from the wiki, everything was working ok when I rebooted the system.
I typed the command: sudo systemctl start mavlink-router then showed the next error:
Job for mavlink-router.service failed because the control process exited with error code. See "systemctl status mavlink-router.service" and "journalctl -xe" for details.

ivanalexo, 2018-04-12:
Thanks for your reply @avinash-palleti,
The problem is with the error `Job for mavlink-router.service failed because the control process exited with error code. See "systemctl status mavlink-router.service" and "journalctl -xe" for details.` How can I correct that? As I mentioned everything was working ok until I rebooted the system and mavlink-router stoped.

ivanalexo, 2018-04-12:
running `journalctl -u mavlink-router` the logs are:
`-- Logs begin at jue 2018-04-12 15:25:11 -04, end at jue 2018-04-12 15:40:57 -04. --
abr 12 15:25:13 intel-rtf-drone systemd[1]: Starting MAVLink Router...
abr 12 15:25:13 intel-rtf-drone bash[753]: /bin/bash: /sys/class/tty/ttyS1/rx_trig_bytes: Permission denied
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Control process exited, code=exited status=1
abr 12 15:25:13 intel-rtf-drone systemd[1]: Failed to start MAVLink Router.
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Unit entered failed state.
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Failed with result 'exit-code'.
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Service hold-off time over, scheduling restart.
abr 12 15:25:13 intel-rtf-drone systemd[1]: Stopped MAVLink Router.
abr 12 15:25:13 intel-rtf-drone systemd[1]: Starting MAVLink Router...
abr 12 15:25:13 intel-rtf-drone bash[835]: /bin/bash: /sys/class/tty/ttyS1/rx_trig_bytes: Permission denied
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Control process exited, code=exited status=1
abr 12 15:25:13 intel-rtf-drone systemd[1]: Failed to start MAVLink Router.
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Unit entered failed state.
abr 12 15:25:13 intel-rtf-drone systemd[1]: mavlink-router.service: Failed with result 'exit-code'.
abr 12 15:25:14 intel-rtf-drone systemd[1]: mavlink-router.service: Service hold-off time over, scheduling restart.
abr 12 15:25:14 intel-rtf-drone systemd[1]: Stopped MAVLink Router.
abr 12 15:25:14 intel-rtf-drone systemd[1]: Starting MAVLink Router...
abr 12 15:25:14 intel-rtf-drone bash[872]: /bin/bash: /sys/class/tty/ttyS1/rx_trig_bytes: Permission denied
abr 12 15:25:14 intel-rtf-drone systemd[1]: mavlink-router.service: Control process exited, code=exited status=1
abr 12 15:25:14 intel-rtf-drone systemd[1]: Failed to start MAVLink Router.
abr 12 15:25:14 intel-rtf-drone systemd[1]: mavlink-router.service: Unit entered failed state.
abr 12 15:25:14 intel-rtf-drone systemd[1]: mavlink-router.service: Failed with result 'exit-code'.
`
@avinash-palleti

zehortigoza, 2018-04-13:
Check if you have booted with aero kernel.
Are you running the line to start the mavlink-router with root, right?

avinash-palleti, 2018-04-14:
If you have installed aero-system package, system will boot automatically with aero kernel.
Can you provide output for the below command,
`dpkg -s aero-system`

closed 2018-06-05 by zehortigoza