intel-aero/meta-intel-aero #339: Need newest manifest for intel-aero version 1.6.1, and couple of more things for better understanding?!

ZoranStojsavljevic, 2018-04-21:
Hello to INTEL intel-aero maintainers,

I am doing some experiments with this project, actually trying to use ATOM Braswell based platform.

I would like to know the following answers to my questions, if you allow me?
_______
[1] You are talking about the pre-built YOCTO environment, as one of the (two) solutions?!
      What would be the manifest to rebuild the latest whole intel-aero 1.6.1 project (I would like to be in control of the source code, I do not like too much pre-built environment)?
      I assume the following for the newest 1.6.1 manifest:
      repo init -u https://github.com/intel-aero/intel-aero-manifest.git -m v1.6.1.xml -b refs/tags/v1.6.1
[2] You are talking about replacement (as my best understanding is) to YOCTO 1.6.1 (I assume it is still Krogoth based, correct me if I am wrong), which is Ubuntu 16,04, with built on top ROS (Robotic OS)?
      Am I correct?
[3] It seems that interfaces for both projects are very similar/the same, so all the intel-areo specific repos and projects are API compliant to both intel-areo 1.6.1 and Ubuntu 16.04?
     Am I correct?
[4] What are pros, and what cons to use YOCTO 1.6.1 versus Ubuntu 16.04 release?
     Can you, please, outline specifics in detail?
[5] Do you plan to upgrade YOCTO Krogoth release to Rocko (2.4.x), or YOCTO 2.5, or 2.6 releases?
     If yes, what is the time line for this to happen?
[6]  Do you plan to upgrade Ubuntu 16.04 release to Ubuntu 17.10, or 18.04 releases?
     If yes, what is the time line for this to happen?
[7] Do you plan to release some new CORE platform for intel-aero project in the near Future?
_______
I would like to thank you for the answers,
Zoran Stojsavljevic

avinash-palleti, 2018-04-22:
[1] Yes, manifest file you referred is right one for getting Intel Aero 1.6.1 source.
[2] [3] [4] Aero packages on top of Ubuntu 16.04 are not replacement to Yocto 1.6.1, that is alternative to Yocto build. There are no differences in both the images, both of them has similar functionalities with respect to Intel Aero. Ubuntu release is made for the people who are not comfortable with Yocto environment.
[5][6][7] As of now there are no plans to upgrade to Rocko, 17.10 or to new CORE platform.

ZoranStojsavljevic, 2018-04-24:
OK, I did some more investigation. Here is the 1.6.1 build configuration, all built and linked by me, in order to investigate further:
https://pastebin.com/pvBhaDYm

As I read from the above, the following is true (you re-based the whole project from Krogoth at 1.0.0 to Pyro 1.6.1):
[1] Poky is true based on pyro-17.0.3, which means that you achieved the effort to make very stable basis for INTEL-AERO: Pyro version 2.3.3, stable release! Kudos to you, since Pyro is still maintained YOCTO version;
[2] The latest three layers are also advanced, you advanced some packets for RealSence 200 (3D) Camera;
[3] MAVLINK protocol is IN, and all the relevant packages;
[4] The whole meta-intel-aero project is on 1.6.1 branch!

QUESTIONS:
[1] As I understand, this is enough to use all the products from intel-aero/build/tmp/deploy/images/intel-aero/ seamlessly. But, BIOS (UEFI actually, comes separately, out of INTEL-AERO YOCTO, am I correct)?
[2] If I go this route, this is all I need to run all other stuff (card programming and drone adjusting), as it is presented in (+ BIOS separately):
https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-Setup ?
[3] Do I need o make .iso from INTEL-AERO also? How I can achieve that (part which is missing in [1] and [2])???
[4] As I understood, the flight application is also in there?! Which layer is flight application in? meta-intel-aero? And/Or?
[5] I see that you communicate to MAVLINK protocol (which is part of meta-intel-aero?!) via serial? I assume, this is serial over Wi-Fi, correct?

Thank you,
Zoran