intel-aero/meta-intel-aero #344: M.2 USB 3.0 Host Controller

jiajun78, 2018-05-04:
Hi,

My setup is as follows:
OS: Ubuntu 16.04
Kernel: 4.4.76-aero-1.2

I am trying to integrate a M.2 B+M key Renasas uPD720202 USB 3.0 Host Controller into the M.2 SSD slot. The problem is the Intel Aero Compute will not detect this Renasas PCI Express device when I typed "lspci" in the terminal. Is there any BIOS setting that Intel is setting that does not allow PCI Express configuration cycle to take place for non-SSD device?

The reason why I need to expand the USB ports of Intel Aero Compute is because I have used/configured the existing USB port as a USB end-point/slave and I can't hook up any additional USB devices (such as Intel Movidius) that require host controller.

Can Intel remove the restriction that only M.2 SSD is supported and allow other non-SSD M.2 device to be connected to the Intel Aero Compute Board?

zehortigoza, 2018-05-04:
We were able to connect and use a LTE modem(Intel’s​ ​ XMM​ ​ 7160​) in that port, did you upgraded the BIOS? To enable it there is a change in the BIOS of release v1.5.0 to do so.

jiajun78, 2018-05-07:
The LTE modem that you are refering to is residing in another M.2 port that has a USB connection. What I am refering to the M.2 Key M 2280 SSD port where there is a PCI express connection.

For your information, we are already using the latest BIOS version as we do have a LTE modem installed in our board configuration.

jiajun78, 2018-05-11:
Issue is solved. Thanks

closed 2018-05-11 by jiajun78