intel-aero/meta-intel-aero #345: Complete power failure upon switch to position control mode

rallen10, 2018-05-04:
Also posted here: https://communities.intel.com/message/541263

I have just come across seemingly major issue while working with the Ready-to-Fly. I am attempting to get the PX4 position control flight mode working on the RTF and it's causing an instantaneous power shutdown for the entire drone (not just the Aero Flight Controller, the entire system)!

For some context I am trying to use local_position_estimator app from PX4. Folks I have spoken to in other research groups have had continued problems with the ekf2 app on PX4 so I am trying a different route using local_position_estimator.

Here is my setup and workflow

Aero Compute Board:

- running Ubuntu 16.04

- setup as desktop workstation, i.e. props removed, wall power as opposed to battery, monitor+keyboard+mouse interface

Aero Flight Controller:

- running the px4 build that comes with the drone: /etc/aerofc/px4/nuttx-aerofc-v1-default.px4

- (note same result if I flash the most recent commit to master on https://github.com/PX4/Firmware)

Workflow:

1. From Aero Compute Board, access mavlink shell via Aero Compute Board

    ```
    cd ~/Firmware/Tools
    python mavlink_shell.py 0.0.0.0:14550
    ```

2. shutdown ekf2 and startup attitude_estimator_q and local_position_estimator
    ```
    ekf2 stop
    attitude_estimator_q start
    local_position_estimator start
    ```

3. Manually arm quadrotor and attempt to switch to position control mode

RESULT:

+ The entire Intel RTF completely powers down almost instantaneously; not just AeroFC, the entire drone complete powers off!!.
+ I will note that it has just enough time to send a REJECT POSITION CONTROL message to QGroundControl.
+ This does NOT happen if the quad is NOT armed

While it might be easy to disregard this as a PX4 problem, I disagree. Whatever PX4 is doing, it shouldn't be able to cause a complete power failure for the Intel RTF platform. Furthermore I am using a "vanilla" version of PX4 so it's not doing anything exotic.

zehortigoza, 2018-05-04:
It also happens in other modes and with EKF2? I guess it is a problem with your battery, it is not being able to supply enough current. It should happen in any mode with any estimator.
And you are right AeroFC has no relation with power control in Aero RTF, so it is something else causing this.

lucasdemarchi, 2018-05-08:
> setup as desktop workstation, i.e. props removed, wall power as opposed to battery, monitor+keyboard+mouse interface

When you arm you are probably drawing more power than your power supply is capably of delivering, voltage drops and board shuts down.  I don't think there's any relation to the estimator you are using.  Switch to a battery and see if problem persists.

closed 2018-06-05 by zehortigoza

rallen10, 2018-06-26:
Sorry for not following up here, I was also tracking this problem here: https://communities.intel.com/message/541263