intel-aero/meta-intel-aero #348: Do anyone use the drone for visual-inertial odometey/slam research?

zxp771, 2018-05-16:
I‘m wondering how can I get the GPS data and IMU date from the drone.
I want use the drone to do some development and research in visual inertial odometry/slam.
These data is very important and it better to get these data and the video data in time synchronization condition.

zehortigoza, 2018-05-16:
MAVLink, check examples here: https://github.com/intel/mavlink-router/tree/master/examples
Just modify to get the messages that you want: http://mavlink.org/messages/common

zxp771, 2018-05-22:
Hi @zehortigoza 

Thanks for your help.
I know how to get those data now it's easy by ros.
But I still have some question.
1.Could I change the imu operating frequency by myself.I record the imu/data it seems like its operating frequency is 50hz. I'm wondering if the imu can reach 100 or 200hz. because for most visual-inertial algorithms the data should be collect in this frequency. 
2.how can I do the extrinsic parameter calibration(camera between imu)?
3.Does the imu and camera data I get from ros should in synchronization condition?I use rosbag fuction to record these data.

Thanks for your help again.

zxp771, 2018-06-20:
Hi @szebedy 

I tried both of the methods you told me but none of them work.
Do you have some new idea?
BTW have you done the extrinsic calibration between the imu and camera?

zxp771, 2018-06-20:
Hi @szebedy 
I just make it by use this command in terminal"rosrun mavros mavcmd long 511 105 10000 0 0 0 0 0"
to boom up the imu rate to 100hz.but actually I don't know the meaning of "10000 0 0 0 0 0"
I want to boom up the rate to 200hz for the extrinsic calibration.（I use the software which named kalibr: https://github.com/ethz-asl/kalibr/wiki/calibrating-the-vi-sensor）

lbegani, 2018-06-21:
> Do you try other visual/visual-inertial algorithm like ORB-SLAM,VINS-mono?

@zxp771 I have tried running VINS-MONO. I have used bottom facing camera and BMI160 IMU sensor. The output was visualized in Rviz. I am yet to integrate it with MAVROS to send MAVLink messages to flight stack.
VINS-MONO does auto calibration for IMU-Camera extrinsic parameters.

zxp771, 2018-06-21:
Hi@lbegani 
Thanks for your reply.
How is the performance of the VINS-mono?

zxp771, 2018-06-21:
Hi @szebedy 
Thanks for your help.
I'm trying to use the Kalibr to do the calibration.
I will share the result to you when I finish.

szebedy, 2018-06-21:
Yes it does

lbegani, 2018-06-22:
```
Hi@lbegani
Thanks for your reply.
How is the performance of the VINS-mono?
```

@zxp771 The performance(in Aero) in terms of pose estimation accuracy looks good. However in terms of processing time taken, its not looking good. I have disabled loop closure and need to do few more optimizations to bring down the latency.
If you are looking for performance metrics, there is a paper on benchmark comparison between different open source VIO algos - http://rpg.ifi.uzh.ch/docs/ICRA18_Delmerico.pdf

zxp771, 2018-06-22:
Hi @lbegani 
Thanks for your reply and sharing.
I have read this paper before.

araujokth, 2018-10-25:
Hi everyone,

I was reading this thread and could not get a conclusion if the aero platform allowed the capturing of measurements of IMU and the Aero cameras in synch? Has anyone verified how well the synch between these measurements is? 

Also, can we synch the IMU measurements with the camera measurements from any of the 3 cameras in the Aero (real sense, 8MP front facing and VGA bottom facing)?

moshanATucsd, 2018-12-11:
Hi, does anyone know how to capture synced imu and camera data from aero?

zxp771, 2018-12-27:
@moshanATucsd 
Hi, As I know the imu and camera data on the drone need to do some post-processing to get the synced data. The Drone won't give us these directly.

zxp771, 2018-12-27:
Hi @araujokth 
Unfortunately, the drone won't give us the synced data directly. We only can do post-processing with these data.

szebedy, 2019-01-01:
Hi @zxp771,

Unfortunately I don't have any of the datasets anymore. But you can check out my repository written for the Intel drone under [this link](https://github.com/szebedy/autonomous-drone). I might have some useful information about visual inertial odometry.

zxp771, 2019-01-02:
Hi @szebedy 
Thanks for your sharing!
It's really helpful!