intel-aero/meta-intel-aero #350: What is the best configuration for the Intel RealSense camera to get a good video ?

RohanKurane, 2018-05-22:
Hello,

I am struggling to find the best configuration for the Intel RealSense camera to get good video quality. I see there are a bunch of options that can be set for the camera. Can you suggest the best options to set here for the best possible video quality ?

brightness (int)    : min=0 max=255 step=1 default=56 value=56
                      contrast (int)    : min=16 max=64 step=1 default=32 value=32
                    saturation (int)    : min=0 max=255 step=1 default=128 value=128
                           hue (int)    : min=-2200 max=2200 step=1 default=0 value=0
white_balance_temperature_auto (bool)   : default=1 value=1
                         gamma (int)    : min=100 max=280 step=1 default=220 value=220
                          gain (int)    : min=0 max=256 step=1 default=32 value=32
          power_line_frequency (menu)   : min=0 max=2 default=0 value=0
     white_balance_temperature (int)    : min=2000 max=8000 step=1 default=6500 value=6500 flags=inactive
                     sharpness (int)    : min=0 max=7 step=1 default=0 value=0
        backlight_compensation (int)    : min=0 max=4 step=1 default=1 value=1
                 exposure_auto (menu)   : min=0 max=3 default=3 value=3
             exposure_absolute (int)    : min=1 max=666 step=1 default=1 value=1 flags=inactive

lbegani, 2018-05-24:
IMO good video quality is subjective. Use the default values if you are not sure about the parameters.

RohanKurane, 2018-05-24:
The defaults had a dark video quality. I tried increasing the brightness to 180 but still I think it could do better. What about contrast or backlight compensation for outdoor video ?

lbegani, 2018-05-25:
Try playing with the values and use the ones that gives best result to you. Unfortunately I do not have something like recommended list of values for the parameters.

closed 2019-06-25 by RohanKurane