intel-aero/meta-intel-aero #351: System Time and Hardware Time won't keep after reboot

trgiman, 2018-05-25:
Hi all,

1.I tried to setup hardware time in bios. Than I did continue with boot cycle. Than called timedatectl and values were correct.

2.Than I powered off the board. Than I booted Yocto again and. timedatectl is showing wrong values.

3.Any suggestions please how to keep the correct time without Network time on? Or does it mean, that Aero Board does not have Real Time Circuit? 

Or should I modify rcS scripts somewhere?

trgiman, 2018-05-29:
Hi @avinash-palleti I noticed you released new bios updat. I am wondering can it help with issue, that Aeroboard can not maintain correct time after disconnected from power or it is missing real time clock circuit that maintaines actual time? My comments in previous comment. Could you help to clarify it please?

zehortigoza, 2018-05-29:
Other option would be use the time received from GPS satellites

trgiman, 2018-05-29:
@zehortigoza @broody thanks for confirmation.Got it. It is for indoor operations so we will use WIFI than. In case, that there would be some expansion RTC module like for Raspberry PI it would be interesting. I think we can close the issue.

closed 2018-05-29 by trgiman