intel-aero/meta-intel-aero #352: /sys/firmware/dmi/tables/smbios_entry_point: Permission denied /dev/mem: Permission denied

zxp771, 2018-05-30:
There is something wrong with the BIOS.
I have followed all the installation on Wiki.
In fact when I finshed the installation I use the aero-get-version.py to check all the state and it was correct like this:
BIOS_VERSION = Aero-01.00.16
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.6.5

But after several times fiy I found the drone got hard to handle.
so I checked the firmware by aero-get-version.py today and the result is here:

/sys/firmware/dmi/tables/smbios_entry_point: Permission denied
/dev/mem: Permission denied
BIOS_VERSION = unknown
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = unknown
AeroFC firmware version = 1.6.5

I don't know what happened.
I try to reflash the BIOS as the link: https://communities.intel.com/thread/124337
but it didn't work.

aero@aero-CherryTrail:~$ aero-get-version.py

/sys/firmware/dmi/tables/smbios_entry_point: Permission denied
/dev/mem: Permission denied
BIOS_VERSION = unknown
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = unknown
AeroFC firmware version = 1.6.5

aero@aero-CherryTrail:~$ dpkg -s aero-system
Package: aero-system
Status: install ok installed
Priority: important
Section: misc
Installed-Size: 9
Maintainer: Daniel E Ruano <daniel.e.ruano@intel.com>
Architecture: all
Multi-Arch: foreign
Version: 1.0.10
Depends: linux-image-4.4.76-aero-1.2, linux-headers-4.4.76-aero-1.2, aero-mavlink-router, aero-fpga, aero-px4, aero-ardupilot, aero-utils, aero-init, aero-bios, openssh-server, i2c-tools, firmware-atomisp, aero-systemd, aero-camera-streaming-daemon, aero-optical-flow
Description: Collection of Intel Aero packages.
 This collection of Intel Aero packages includes all necessary programs and
 utilities to use the Intel Aero RTF drone on Ubuntu.
Homepage: https://github.com/intel-aero/

aero@aero-CherryTrail:~$ uname -r
4.4.76-aero-1.2

lucasdemarchi, 2018-05-30:
what are the permission bits on the 2 files it can't access?

zxp771, 2018-05-31:
Hi @lucasdemarchi 

Thanks for your reply.
I don't know what the permission is.
I have never meet these problem before.
I just use the command" aero-get-version.py "to check the firmware state.

zxp771, 2018-05-31:
Hi @lucasdemarchi 

Thanks for your help.
I just figure out.
I should use sudo before the " aero-get-version.py "
I'll close this problem.

closed 2018-05-31 by zxp771