intel-aero/meta-intel-aero #353: mavlink-router broke when armed

Huibean, 2018-05-31:
every time when arm the drone, mavlink connection broke and mavlink router show errors
` mavlink-routerd[9775]: Open UDP [4] 192.168.8.255:14550
 mavlink-routerd[9775]: Open UDP [5] 192.168.7.255:14550
mavlink-routerd[9775]: Open UDP [6] 0.0.0.0:80550 *
mavlink-routerd[9775]: Open UART [7] /dev/ttyS1 *
mavlink-routerd[9775]: UART [7] speed = 921600
mavlink-routerd[9775]: Open TCP [9] 0.0.0.0:5760 *
mavlink-routerd[9775]: Baudrate 921600 responded, keeping it
mavlink-routerd[9775]: Logging target system_id=1 on 00015-2018-05-31_03-25-03.bin
systemd[1]: mavlink-router.service: Main process exited, code=dumped, status=11/SEGV
systemd[1]: mavlink-router.service: Unit entered failed state.
systemd[1]: mavlink-router.service: Failed with result 'core-dump'.
systemd[1]: mavlink-router.service: Service hold-off time over, scheduling restart.
systemd[1]: Stopped MAVLink Router.
systemd[1]: mavlink-router.service: Start request repeated too quickly.
systemd[1]: Failed to start MAVLink Router.`

lucasdemarchi, 2018-05-31:
Do you have the last versions? PX4 or Ardupilot?

Please run this:

```
systemctl stop mavlink-router
mavlink-routerd -v
```

And do you have changes to its configuration file?

Huibean, 2018-05-31:
@lucasdemarchi can not upgrade bios in ubuntu, I just flash yocto and upgrade bios to 01.00.16 and test mavlink connection while armed it works fine, just now I flash ubuntu again and test, it works fine now, thanks

closed 2018-06-01 by Huibean