intel-aero/meta-intel-aero #354: QGroundControl as X11 Client XTerm UBUNTU Application

ZoranStojsavljevic, 2018-05-31:
I found the following net pointer:
https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/apu-170405-pixhawk-flight-guide.pdf

From it, I have spotted something which I'll really need. I am wondering, since I have QGroundControl as application ONLY on WIN10, but from the very same document, I see that the QGroundControl setup menu is (believe or not) presented on UBUNTU's X11 Client XTerm,

Could you, please, direct me where I can find QGroundControl application as:
```
- package to Debian/UBUNTU?
- The same, but just tarball (.tar.bz2)?
- Something else (maybe .rpm or .deb files)?
```

Thank you in advance,
Zoran

lucasdemarchi, 2018-05-31:
Their website should answer your question. QGC is distributed for Linux as an .appimg that you can run. And it may be packaged by some distros. 
https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html

ZoranStojsavljevic, 2018-05-31:
> download_and_install.html

I found it as well:
https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html

Thank you,
Zoran

closed 2018-05-31 by ZoranStojsavljevic