intel-aero/meta-intel-aero #355: is aero support Ardupilot 3.6 now ?

Huibean, 2018-06-01:
after upload firmware build form newest master branch, it lost connect with flight controller, and firmware version show unknown, directly flash firmware from http://firmware.ardupilot.org/Copter/latest/aerofc-v1/ get the same result

```
huibean@huibean-CherryTrail:~$ sudo aero-get-version.py
BIOS_VERSION = Aero-01.00.16
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown
```

here is logs from ./waf copter --upload

```
➜  ardupilot git:(master) ./waf copter --upload
Waf: Entering directory `/Users/huibean/Work/ardupilot/build/aerofc-v1'
[ 7/16] CMake Configure px4
-- nuttx-aerofc-v1-apm
-- could not find gdbtui
-- could not find ddd
-- could not find gdbtui
-- could not find ddd
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Found PythonInterp: /Users/huibean/anaconda3/bin/python (found version "3.6.3")
-- GIT_DESC = b535f9744f74d537580f3d396ba01a6f31d917ae
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware
INCLUDE_DIRS=/Users/huibean/Work/ardupilot/modules/PX4Firmware/src;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/modules;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/include;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/lib;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/platforms;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/drivers/boards/aerofc-v1;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src/modules/px4_messages;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src/modules;/Users/huibean/Work/ardupilot/modules/PX4Firmware/mavlink/include/mavlink;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/lib/DriverFramework/framework/include;src/lib/matrix;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/include;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/include/cxx;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/arch/chip;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/arch/common
ADDING DRIVERS

[15/16] CMake Build px4 msg_gen
ninja: no work to do.
[16/16] CMake Build px4 prebuild_targets
ninja: no work to do.
[480/484] CMake Configure px4
-- nuttx-aerofc-v1-apm
-- could not find gdbtui
-- could not find ddd
-- could not find gdbtui
-- could not find ddd
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Found PythonInterp: /Users/huibean/anaconda3/bin/python (found version "3.6.3")
-- GIT_DESC = b535f9744f74d537580f3d396ba01a6f31d917ae
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware
INCLUDE_DIRS=/Users/huibean/Work/ardupilot/modules/PX4Firmware/src;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/modules;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/include;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/lib;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/platforms;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/drivers/boards/aerofc-v1;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src/modules/px4_messages;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/src/modules;/Users/huibean/Work/ardupilot/modules/PX4Firmware/mavlink/include/mavlink;/Users/huibean/Work/ardupilot/modules/PX4Firmware/src/lib/DriverFramework/framework/include;src/lib/matrix;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/include;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/include/cxx;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/arch/chip;/Users/huibean/Work/ardupilot/build/aerofc-v1/modules/PX4Firmware/aerofc-v1/NuttX/nuttx-export/arch/common
ADDING DRIVERS

[481/484] CMake Build px4 build_firmware_aerofc-v1
ninja: no work to do.
[482/484] CMake Build px4 upload
[0/1] uploading /Users/huibean/Work/ar...firmware/nuttx/nuttx-aerofc-v1-apm.px4
Copying files to Aero board (root@intel-aero.local)...
nuttx-aerofc-v1-apm.px4                       100%  807KB  14.7MB/s   00:00
px_uploader.py                                100%   30KB   7.5MB/s   00:00
Linux huibean-CherryTrail 4.4.76-aero-1.2 #1 SMP PREEMPT Thu Dec 21 09:55:38 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

BIOS_VERSION = Aero-01.00.16
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = 3.5.3

Updating firmware on AeroFC
==========================================================================================================
WARNING: You should uninstall ModemManager as it conflicts with any non-modem serial device (like Pixhawk)
==========================================================================================================
Loaded firmware for 41,0, size: 911504 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board 41,0 bootloader rev 5 on /dev/ttyS1
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff
Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

Firmware updated
Restarting router
Finished.
/dev/ttyS1:
Attempting reboot on /dev/ttyS1 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=460800...
If the board does not respond, unplug and re-plug the USB connector.

Waf: Leaving directory `/Users/huibean/Work/ardupilot/build/aerofc-v1'

BUILD SUMMARY
Build directory: /Users/huibean/Work/ardupilot/build/aerofc-v1
Target          Text    Data  BSS    Total
-------------------------------------------
bin/arducopter  909424  2080  58740  970244

PX4
The ELF files are pointed by the path in the "Target" column. The .px4 files are in
the same directory of their corresponding ELF files.

Build commands will be stored in build/aerofc-v1/compile_commands.json
'copter' finished successfully (1m53.170s)
```

trgiman, 2018-06-04:
hi @Huibean unfortunately I have same result... however ardupilot copter 3.5.5 compiled on my Linux machine works correctly

lucasdemarchi, 2018-06-04:
There was a regression on ardupilot on master branch. I suggest you to follow the upstream issue: https://github.com/ArduPilot/ardupilot/issues/8416

I did some quick debug, but it's not working yet.

Huibean, 2018-06-05:
@lucasdemarchi ok, get it

closed 2018-06-05 by zehortigoza