intel-aero/meta-intel-aero #356: dev/spidev* not found

abhi11796acharya, 2018-06-01:
Hello,

I installed latest version of yocto and everything was working fine. I was able to access all I2C and SPI buses. But after installing ubuntu, I am not able to access SPI buses. Whenever I try to the below mentioned command on shell, it throws an error saying "dev/spidev3.0 not found".

spi_xfer -b 3 -c 0 -d 0x80 -w 2

I followed the wiki page procedure to install ubuntu.
Kindly help me!!

Thank you!!

abhi11796acharya, 2018-06-04:
how can I check this ..?

Rjasuja, 2018-06-04:
Run `dpkg -s <package_name>` command

abhi11796acharya, 2018-06-04:
yes it is installed. system is throwing following error:

ERROR: failed to open /dev/spidev3.0

When I searched for spidev3.0 in /dev directory, I didnt find it.

abhi11796acharya, 2018-06-04:
I installed everything again and now spidev1.1 and spidev3.0 is present in the dev directory but I am still getting the same error when trying spi_xfer command on shell.

"ERROR: failed to open /dev/spidev3.0"

abhi11796acharya, 2018-06-04:
I used sudo and now it is working.  :)

Thank you!!

closed 2018-06-04 by abhi11796acharya