intel-aero/meta-intel-aero #357: Do anyone implement the SVO on the aero RTF drone?

zxp771, 2018-06-05:
I'm doing some research on visual odometry and slam.
I just install the svo and using the bottom camera as the input for the algorithm.
but when I run the svo I always got this notes:
SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /svo/cam_cx: 309.936417
 * /svo/cam_cy: 242.427118
 * /svo/cam_d0: -0.022232
 * /svo/cam_d1: -0.034436
 * /svo/cam_d2: 0.003345
 * /svo/cam_d3: 8.5e-05
 * /svo/cam_fx: 414.756208
 * /svo/cam_fy: 414.983046
 * /svo/cam_height: 480
 * /svo/cam_model: Pinhole
 * /svo/cam_topic: /usb_cam/image_raw
 * /svo/cam_width: 640
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

ROS_MASTER_URI=http://localhost:11311

process[svo-1]: started with pid [3429]
create vo_node
[ WARN] [1528234776.616960704]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1528234776.618874699]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1528234776.620884213]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1528234776.625195170]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1528234776.627013647]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1528234776.628746680]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1528234776.832704321]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1528234776.834422304]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1528234776.836315998]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1528234776.838576352]: Found parameter: svo/cam_model, value: Pinhole
[ INFO] [1528234776.840500385]: Found parameter: svo/cam_width, value: 640
[ INFO] [1528234776.842313561]: Found parameter: svo/cam_height, value: 480
[ INFO] [1528234776.844522799]: Found parameter: svo/cam_fx, value: 414.756
[ INFO] [1528234776.846400992]: Found parameter: svo/cam_fy, value: 414.983
[ INFO] [1528234776.848601592]: Found parameter: svo/cam_cx, value: 309.936
[ INFO] [1528234776.850640783]: Found parameter: svo/cam_cy, value: 242.427
[ INFO] [1528234776.852500837]: Found parameter: svo/cam_d0, value: -0.022232
[ INFO] [1528234776.854717613]: Found parameter: svo/cam_d1, value: -0.034436
[ INFO] [1528234776.856453597]: Found parameter: svo/cam_d2, value: 0.003345
[ INFO] [1528234776.858919014]: Found parameter: svo/cam_d3, value: 8.5e-05
[ WARN] [1528234776.913663131]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1528234776.916072582]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1528234776.917951262]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1528234776.919817542]: Found parameter: svo/init_rz, value: 0
[ INFO] [1528234776.921600428]: Found parameter: svo/init_ry, value: 0
[ INFO] [1528234776.923323124]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1528234776.926382753]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1528234776.928838619]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[ WARN] [1528234776.930731638]: Cannot find value for parameter: svo/n_pyr_levels, assigning default: 3
[ WARN] [1528234776.932477822]: Cannot find value for parameter: svo/use_imu, assigning default: 0
[ WARN] [1528234776.934625432]: Cannot find value for parameter: svo/core_n_kfs, assigning default: 3
[ WARN] [1528234776.936946752]: Cannot find value for parameter: svo/map_scale, assigning default: 1
[ INFO] [1528234776.938851797]: Found parameter: svo/grid_size, value: 30
[ WARN] [1528234776.940821370]: Cannot find value for parameter: svo/init_min_disparity, assigning default: 50
[ WARN] [1528234776.942595281]: Cannot find value for parameter: svo/init_min_tracked, assigning default: 50
[ WARN] [1528234776.944481799]: Cannot find value for parameter: svo/init_min_inliers, assigning default: 40
[ WARN] [1528234776.946676124]: Cannot find value for parameter: svo/klt_max_level, assigning default: 4
[ WARN] [1528234776.948418484]: Cannot find value for parameter: svo/klt_min_level, assigning default: 2
[ WARN] [1528234776.950450036]: Cannot find value for parameter: svo/reproj_thresh, assigning default: 2
[ WARN] [1528234776.952328266]: Cannot find value for parameter: svo/poseoptim_thresh, assigning default: 2
[ WARN] [1528234776.954134592]: Cannot find value for parameter: svo/poseoptim_num_iter, assigning default: 10
[ WARN] [1528234776.956149368]: Cannot find value for parameter: svo/structureoptim_max_pts, assigning default: 20
[ WARN] [1528234776.957996159]: Cannot find value for parameter: svo/structureoptim_num_iter, assigning default: 5
[ WARN] [1528234776.959730930]: Cannot find value for parameter: svo/loba_thresh, assigning default: 2
[ WARN] [1528234776.961582308]: Cannot find value for parameter: svo/loba_robust_huber_width, assigning default: 1
[ INFO] [1528234776.963426524]: Found parameter: svo/loba_num_iter, value: 0
[ WARN] [1528234776.965105654]: Cannot find value for parameter: svo/kfselect_mindist, assigning default: 0.12
[ WARN] [1528234776.967127081]: Cannot find value for parameter: svo/triang_min_corner_score, assigning default: 20
[ WARN] [1528234776.969089779]: Cannot find value for parameter: svo/triang_half_patch_size, assigning default: 4
[ WARN] [1528234776.970972647]: Cannot find value for parameter: svo/subpix_n_iter, assigning default: 10
[ INFO] [1528234776.972883866]: Found parameter: svo/max_n_kfs, value: 10
[ WARN] [1528234776.974770797]: Cannot find value for parameter: svo/img_imu_delay, assigning default: 0
[ WARN] [1528234776.976473416]: Cannot find value for parameter: svo/max_fts, assigning default: 120
[ WARN] [1528234776.978373210]: Cannot find value for parameter: svo/quality_min_fts, assigning default: 50
[ WARN] [1528234776.981234977]: Cannot find value for parameter: svo/quality_max_drop_fts, assigning default: 40
[ INFO] [1528234776.981536371]: SVO initialized
[ INFO] [1528234776.988763961]: Found parameter: svo/cam_topic, value: /usb_cam/image_raw
[ INFO] [1528234777.138159246]: RESET
[ WARN] [1528234777.148104982]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234779.173047593]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234781.205711497]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234783.239237902]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234785.272739556]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234787.305729692]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234789.306048594]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234791.339117277]: First image has less than 100 features. Retry in more textured environment.
[ WARN] [1528234793.372534979]: First image has less than 100 features. Retry in more textured environment.

It seems the algorithm was initialized successfully.
I also use rostopic echo /svo/image to check the input.it works well.
but the algorithm wouldn't update any pose information when I move the drone by my hand.

xiaoyaoE, 2018-06-09:
hi, can you please tell me how do you get the camera topic of bottom camera?

zxp771, 2018-06-10:
Hi@xiaoyaoE 
You can read this link :https://communities.intel.com/thread/121658
I think you will find the answer.