intel-aero/meta-intel-aero #359: Can not install zbar with pip install - Python.h is missing in Yocto 1.6.1

trgiman, 2018-06-11:
Hi all. I need to install zbar lib. But can not install it since there is no python-dev package. Any workaround suggestions appreciated - installing with sudo dnf --nogpgcheck uninstall python-devel-2.7.13-25.3.1.x86_64.rpm failed due to 

Error: conflicting requests
  - nothing provides glibc-devel needed by python-devel-2.7.13-25.3.1.x86_64

zehortigoza, 2018-06-11:
Easiest way to solve that is add the packages you need to Yocto build it, then you can copy the rpms to Aero and install.

trgiman, 2018-06-11:
hi @zehortigoza thanks. The problem, that I have is that USB port of Aeroboard is mechanically broken (pins are no longer soldered - somehow it just happened) So I was hoping to do it other-way. I installed dronekit with pip without problems (locally no Internet connection), since it it is distributed as bin, but can not find zbar beeing distributed as wheel format. Anyway I will progress with what you suggested

trgiman, 2018-06-12:
@zehortigoza I solved it other-way.Following approach is not good for production, but for tests and fast fix works well:

1. used zbar python pkg from my x86 ubuntu and copied to python 2.7 pkg location on Yocto Aero (location can be found by using pip show --some python pkg--)
2.copied zbar.so and jpeg.so and libs that zbar requested when used in Python program
3.zbar in Python program is set to run without X11 server

it can save some time especially in my situation. but yocto solution mentioned above is much more systematic. Thanks.