intel-aero/meta-intel-aero #360: use opencv(2.4.13 C++)to access the bottom camera's video stream

zxp771, 2018-06-13:
I'm trying to implement an visual odometry algorithm on the RTF drone(ubuntu16.04).
I meet some problem when I try to access the bottom camera(global shutter) with Opencv API like this:

void BenchmarkNode::runFromFolder()
{

    cv::VideoCapture cap(2);  // open the default camera

    if (!cap.isOpened())  // check if we succeeded
        return ;
    
    
    int img_id = 0;
    for (;;) {

        cv::Mat image;
        cap.read(image);  // get a new frame from camera

        assert(!image.empty());
        img_id++;

        cv::imshow("origin_image", image);
        if (cv::waitKey(1) >= 0) break;
        if(img_id < 100) continue;

        cv::cvtColor(image,image,CV_BGR2GRAY);
        /*
        cv::Mat unimg;
        cam_pinhole_->undistortImage(image,unimg);
        vo_->addImage(unimg, 0.01*img_id);
         */
        vo_->addImage(image, 0.01*img_id);

        // display tracking quality
        if(vo_->lastFrame() != NULL)
        {
            std::cout << "Frame-Id: " << vo_->lastFrame()->id_ << " \t"
                      << "#Features: " << vo_->lastNumObservations() << " \n";
            //<< "Proc. Time: " << vo_->lastProcessingTime()*1000 << "ms \n";
            std::cout<<"Frame pose: "<< vo_->lastFrame()->T_f_w_ <<std::endl;

        }

    }

    cap.release();
    return;

}

but when I want to start the test it will occur this error:
[INFO] SVO initialized
HIGHGUI ERROR: libv4l unable to ioctl S_FMT
Floating point exception (core dumped)

Here is the camera list.
as the list shows I think the bottom camera should be 2.
[image]

I also tried to use:
    cv::VideoCapture cap;
    if(!cap.open("/dev/video2"))
    std::cout<<"camera error"<<std::endl;
to get the video stream.but it will occur this error when i run the program.

GStreamer Plugin: Embedded video playback halted; module source reported: Could not read from resource.
OpenCV Error: Unspecified error (GStreamer: unable to start pipeline
) in cvCaptureFromCAM_GStreamer, file /home/aero/opencv-2.4.13.6/modules/highgui/src/cap_gstreamer.cpp, line 816
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/aero/opencv-2.4.13.6/modules/highgui/src/cap_gstreamer.cpp:816: error: (-2) GStreamer: unable to start pipeline
 in function cvCaptureFromCAM_GStreamer

Aborted (core dumped)

zehortigoza, 2018-06-13:
It is /dev/video2 id=1, the id=1 is important.
Please take a look at the aero-optical-flow code(V4L2), you can check how to open the camera using V4L2 APIs and then try to use other libraries that will call V4L2 APIs for you.

zxp771, 2018-06-13:
@zehortigoza Hi
Thanks for your help.
Do you mean the /capturev4l2/capture.cpp?

zehortigoza, 2018-06-13:
https://github.com/intel-aero/aero-optical-flow/blob/master/src/camera.cpp

zxp771, 2018-06-13:
@zehortigoza 
Thanks for your help.
Do I need install other libraries for V4L2?

zehortigoza, 2018-06-13:
Not sure if your distro has, but you need the V4L2 headers to compile the application.

zxp771, 2018-06-13:
@zehortigoza 
Thanks for your help.
I will try these code but I'm not very familiar with V4L2.
please do not close this issue very soon.
Thanks