intel-aero/meta-intel-aero #361: Accessing BlueTooth on Intel AeroBoard

trgiman, 2018-06-18:
Hi @lucasdemarchi thanks a lot for last help with ArduPilot. According documentation. Intel AeroBoard has both Wifi and Bluetooth (Intel® Dual Band Wireless-AC 8260; 802.11ac, 2x2 MIMO Wi-Fi + Bluetooth adapter), 

but when running  lspci | grep Blue in Yocto it seems there is no BLE device. I would like to use BLE for serial communication with my external devices mounted on RTF.

Additionally output of dmesg | grep Blue in Yocto is 

[    0.532360] Bluetooth: Core ver 2.21
[    0.532387] Bluetooth: HCI device and connection manager initialized
[    0.532396] Bluetooth: HCI socket layer initialized
[    0.532403] Bluetooth: L2CAP socket layer initialized
[    0.532418] Bluetooth: SCO socket layer initialized

Which does not say nothing about available BLE devices.

CAN port is already occupied in standard RTF configuration if I understand correctly.

trgiman, 2018-06-18:
Additional comment. Using

lspci returns:

01:00.0 Network controller: Intel Corporation Wireless 8260 (rev 3a)

rfkill list:

0: phy0: wlan
	Soft blocked: no
	Hard blocked: no

Does it mean BLE driver is missing?

zehortigoza, 2018-06-18:
By a few reasons was chosen to enable Bluetooth, also Aero was not certified by FCC with Bluetooth so it was disabled in hardware side.

trgiman, 2018-06-18:
@zehortigoza so there is no way to enable it? Since same chip is used in Laptops etc. And for these BLE works. Hardware site you mean HW design or just firmware in 8260. 

And if BLE is not the option. Considering Intel RTF. Which port to use for controlling other UART enabled device...maybe only trough WI-FI. USB will be occupied by Movidius?

Thank you.

trgiman, 2018-07-18:
@lucasdemarchi thanks. Actually reason why I am looking for options is, that I want to control lights mounted on Intel RTF. I would like to control them via Intel Aero Board. But from documentation if I understand correctly

1) all PWM outputs are occupied in case of Intel RTF
2) In my design Intel RTF Telemetry (fully occupied by Leddar One and PX4Flow)
3) GPS port used for GPS (but I see 2 wires are not occupied)
4) COMPASS (used I know I can add I2C splitter)
5) USB will be occupied by Movidius 

I am rather new in FPGA. But my understanding from documentation is, that all PWM outputs are occupied for Intel RTF case. Or are there any free PWM or can I change it in VERILOG and solder some additional wires to pins?

Eventually other option is CAN port of AeroBoard. But the documentation is rather incomplete? Thanks for recommendations and ideas.

zehortigoza, 2018-07-18:
You could implement the PWM in FPGA and control it from Atom using SPI but the problem that there is no pin exposed beside the ones on top of Aero RTF.
The 2 pins in the GPS port is disconnected so you can't do anything with it.

And you are also using the M2 slot with the modem right? Someone inserted a M2 to USB card in the M2 slot to gain more one USB port.

trgiman, 2018-07-19:
hi @zehortigoza thanks for suggestions. Very interesting!:

**1. M2 slot** - currently I do not need M2 for modem - do you have got link for card with correct connector?(actually It helps since one of Aeroboards has broken standard USB port, so until heat gun will arrive...) 

**2.Exposed PINs** - currently I control drone only trough Optical Flow (No compass, No gps connected). I would like to evaluate and try PWM in FPGA implementation (use current GPS or compass port) - I saw verilog code but I am quite new to FPGA - any recommendation where to change these ports? On the rest I will have to catch up a little bit.

lucasdemarchi, 2018-07-19:
@trgiman source code for the FPGA firmware is here: https://github.com/intel-aero/intel-aero-fpga . In aero-rtf-kit dir you can see the pins assigned for functionality (see the "set_location_assignment" in https://github.com/intel-aero/intel-aero-fpga/blob/master/aero_rtf_kit/Top.qsf). You will want to assign those pins to your pwm block rather than to aerofc.

There's also a sample for you to start on pwm in https://github.com/intel-aero/intel-aero-fpga/blob/master/aero_sample/rtl/pwm.v:  you will need to extend the current spi slave implementation of the rtf kit in order to allow it to communicate with the CPU.  There are some new [videos from Altera](https://www.youtube.com/playlist?list=PL0pU5hg9yniZjBCoMW4A-vsFt7jYD1Mli) available that may help you getting started. 

A long long time ago I flew Aero board alone using a different spi + pwm block - the ArduPilot driver implementation is this one: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/RCOutput_AeroIO.cpp . Although I don't know where's the source code for that pwm implementation it may serve as inspiration for what you want to do.

lucasdemarchi, 2018-07-19:
@trgiman another way, without involving the FPGA is to check if those pins you are not using can be changed on STM32 to be a pwm channel.  Then you'd need to change the flight stack to use those pins.

zehortigoza, 2018-07-19:
2) Try this one: https://github.com/zehortigoza/verilog-playground/blob/master/src/pwm2.v you just need to integrate it into RTF firmware as Lucas said.

trgiman, 2018-07-20:
@zehortigoza @lucasdemarchi thank you very much for these links and short explanations. I will try them out. Very nice!