intel-aero/meta-intel-aero #362: NO video streaming on ubuntu

hongminy, 2018-06-22:
I bought an Intel Aero RTF drone and I flush it with Ubuntu 16.04 following the instruction. https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation
After that, I could only get HD camera stream on my Qgroundcontrol. Other cameras (rtsp url end with rsir and rsdepth) give me nothing. I could get video normally through the cpp-capture command.

Does anyone experienced the same problem? Any help would be appreciated!

hongminy, 2018-06-22:
intel@intel-aero:~$ ls /dev/video*
/dev/video0   /dev/video11  /dev/video2  /dev/video5  /dev/video8
/dev/video1   /dev/video12  /dev/video3  /dev/video6  /dev/video9
/dev/video10  /dev/video13  /dev/video4  /dev/video7

-----------------------------------------
I could get video devices up to 13

hongminy, 2018-06-22:
intel@intel-aero:~$ systemctl status csd
● csd.service - Camera Streaming Daemon
   Loaded: loaded (/lib/systemd/system/csd.service; enabled; vendor preset: enab
   Active: active (running) since Fri 2018-06-22 00:15:57 PDT; 16min ago
 Main PID: 2339 (csd)
   CGroup: /system.slice/csd.service
           └─2339 /usr/bin/csd

Warning: Journal has been rotated since unit was started. Log output is incomple
lines 1-8/8 (END)
-----------------------------------------------------------------------------
The intel realsense service is running

lbegani, 2018-06-22:
@hongminy Can you check the details given in this link - https://github.com/intel-aero/meta-intel-aero/wiki/06-Cameras-and-Video

hongminy, 2018-06-22:
@lbegani I went through the instruction again and it still does not work. 
rtsp://AERO_IP_ON_YOUR_NETWORK:8554/video13 this work fine on my Qgroundcontrol
------------------------
but I can get video through the wifi on the rest 4 feeds
RealSense R200, depth sensor: rtsp://AERO_IP_ON_YOUR_NETWORK:8554/rsdepth
RealSense R200, infrared first camera: rtsp://AERO_IP_ON_YOUR_NETWORK:8554/rsir
RealSense R200, infrared second camera: rtsp://AERO_IP_ON_YOUR_NETWORK:8554/rsir2
Bottom facing black and white global shutter: rtsp://AERO_IP_ON_YOUR_NETWORK:8554/bottom

lbegani, 2018-06-25:
@hongminy If you do not see error logs, you may have to debug the individual cameras to see if they work.  Try to capture from bottom camera using the [steps ](https://github.com/intel-aero/meta-intel-aero/wiki/06-Cameras-and-Video#take-a-photo-with-the-black-and-white-camera) and validate that you are getting camera frames from the camera sensor.