intel-aero/meta-intel-aero #363: running ubuntu 16.04.04, usb keyboard no longer works even on boot

studentofchange, 2018-06-28:
Waiming Mok     Waiming Mok Jun 28, 2018 11:13 AM
I upgraded the aero board to ubuntu 16.04.04 using usb install. Ubuntu desktop comes up on HDMI, but keyboard and other USB devices no longer accessible over OTG USB.

- connect USB hub to OTG USB port, light on USB hub lights up, but keyboard connected to hub no working, neither does mouse
- connect USB keyboard directrly to OTG USB, does not work

I can access ubuntu over the Wifi AP, but no access to internet to do updates.

1. How do I turn on USB when running Ubuntu (USB keyboard was running before ubuntu install)?

     Tried rebooting system,but could not go into BIOS as keyboard does not work.

2. Could I access system (BIOS, OS) using serial console with UART on power connector on Aero board?

3. Is there another way to fix this?

Thank you,
Waiming

zehortigoza, 2018-06-29:
If it do work even in bios it is some problem in your hardware setup.

zehortigoza, 2018-06-29:
It is this one that makes UART free for Atom: https://github.com/intel-aero/meta-intel-aero/blob/master/recipes-support/jam-stapl/jam-stapl/aero-compute-board.jam

AlexisTM, 2018-08-22:
I have the same issue: Mouse and keyboard works fine in the BIOS and during installation, but once the 16.04.4 is installed the keyboard and mouse are no more usable.

I will try the 16.04.3 which was working before.

AlexisTM, 2018-08-23:
I reinstalled the drone using Ubuntu 16.04.3 but the same issue was there.

The issue seems to be kernel related. 

As the keyboard is available from keyboard onwards until the Kernel reload I can go into the advanced boot options of GRUB.

It seems that the installer installed the kernel 4.15.x while the base kernel is 4.10.x. Therefore, I booted up using the Kernel 4.10.x and it worked fine.

AlexisTM, 2018-09-05:
NOTE: After installing aero-system; I removed any other kernel than the **4.4.76-aero-1.2** provided to ensure compatibility.

lucasdemarchi, 2018-09-05:
I'm closing this since a solution was found: use the provided kernel rather than Ubuntu's default

HoaPham2711, 2018-10-26:
> I'm closing this since a solution was found: use the provided kernel rather than Ubuntu's default

Can you give me a detailed guide to resolve this problem? Please help me!

AlexisTM, 2018-10-26:
@HoaPham2711 
Change the kernel used in GRUB edit section

HoaPham2711, 2018-10-27:
I try to press and hold "Shift" key in during boot time but the GRUB menu doesn't appear

HoaPham2711, 2018-10-27:
Seems that USB is completely disabled so my keyboard is unavailable

AlexisTM, 2018-10-27:
Change the keyboard.
This bug is only valid in Linux not during pre-boot.

AlexisTM, 2018-10-27:
Then I really have no way to help you :/

AlexisTM, 2018-10-28:
@HoaPham2711 

Here it is ;) 

https://github.com/intel-aero/meta-intel-aero/issues/279

You cannot use the keyboard but can use the remote connection obviously ;)

HoaPham2711, 2018-10-29:
Thank you but it still doesn't work for me

HoaPham2711, 2018-10-29:
> or.... you disabled xhci support in a bios option, which basically locks you out of USB access. You can access the bios to revert that back using the UART on the power connector, but first you will need to program the FPGA to disconnect that UART from the flight controller (if you are using the RTF) or physically disconnect it (not recommended if you never disassembled it). @zehortigoza do we have a .jam file with that done already?

Can you give me a detailed guide? Please help me :(

sinbadflyce, 2018-10-30:
Hi all,
I got same problem (cannot use the mouse, keyboard and ssh) after upgraded BIOS, installed Ubuntu 16.04.3. Right now, I cannot do anymore with my Intel Aero RTF. I am thinking about the way to hardware reset BIOS, but I cannot how do that. Very thank for any your help.

AlexisTM, 2018-10-31:
I have no clue, I never did it. But make sure to document the procedure to help the others in your condition

lucasdemarchi, 2018-10-31:
You will need to flash the fpga with aero-compute-board.jam. See https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup#flashing-the-fpga

Do you have access to the UART pins on the power connector? See section 3.6.1 of https://software.intel.com/sites/default/files/managed/25/d5/Intel-Aero-Compute-Board-Getting-Started.pdf - You need a usb-serial to connect to your computer and connect to the pins. You can follow https://github.com/intel-aero/meta-intel-aero/wiki/Connecting-an-External-Flight-Controller but instead of using a cable to connect to an external flight controller, you can connect to a usb-serial.

sinbadflyce, 2018-11-01:
Hi Lucas,
I am using Intel Aero RTF Drone. I had already flashed FPGA before I upgraded to Ubuntu 16.0.4.03 Desktop. Then, I got this problem. So, do I still need to re-flash FPGA? Can I flash FPGA with UART on power connector?
Thank you,

lucasdemarchi, 2018-11-01:
You have to flash a different fpga image to "disconnect" the internal flight controller from the uart port. So yes.

And yes for the second question, top

sinbadflyce, 2018-11-02:
Thank you @lucasdemarchi

githublearner-li, 2019-08-23:
hi everyone ！i am a greenhand, can you tell how to change the kernel used in GRUB edit section or give me a detailed guidance ? is this procedure done behind installing the ubuntu system or before?