intel-aero/meta-intel-aero #364: "light" Git version in yocto does not support submodules

jonathan84clark, 2018-07-03:
The version of Git on the intel-aero drone out of the box does not support submodules. There are only two ways around the issue.
1. Re-build Git from source (which has many external dependencies that need to be chased down)
2. Pull the submodules in on a different machine then SCP into the Intel drone and place the submodule code where it should be (which works great unless the submodules are ever updated)

zehortigoza, 2018-07-03:
You can also build Yocto image adding git-perltools that will add submodule for you and then install the rpm or flash the whole image.

jonathan84clark, 2018-07-03:
Thanks for the help!

lucasdemarchi, 2018-07-03:
There are many alternatives... Developing in your computer and copying the result to Aero avoids much of the hassle of needing the tools there. If you want to use Aero as the development machine rather than the target then you need to 1) use Ubuntu (either removing yocto or as a container), or 2) generate the  yoco image yourself with required tools or 3) Compile the dependencies by hand.

closed 2018-07-03 by lucasdemarchi