intel-aero/meta-intel-aero #365: Intel aero damage after crash

talma, 2018-07-10:
Hi,
We crash intel aero drone. When try to turn on the drone it is immediately turned off after some lights turned on. There is a rotor leds (two whites and two red) that may be damaged: one of the red leds is off. 
The GSP stick was also damaged. 
Do we need to replace the rotor or the attached board ?
Can you explain how to verify that GSP is fine? 

Thanks in advance
Tal

zehortigoza, 2018-07-10:
If the ESC led is not blinking it can be a problem in the ESC or in the wires. Other good test to do is arm the drone without the propellers attached and check if the motors are spinning.
You can check if the GPS is working by turning the drone on in a outdoor area, you don't need to arm the drone and check the location in QGC.
You can buy replacement parts here: http://us.yuneec.com/intelaerodroneparts
Keep in mind that 2 motors should spin clockwise and 2 motors should spin counter clock wise you can google to check what is the rotation of each motors.

lucasdemarchi, 2018-07-10:
The problem of not turning on is probably due to a faulty ESC. Disconnect all the ESCs and try to turn it on again - if it doesn't turn on, then you have bigger problems. If it does, find the faulty one by connecting one at a time, and turning the drone on.

You can disconnect the ESC by unplugging the cable that arrives on it or by opening the PDB (where's the power on button) and disconnect them there.

talma, 2018-07-11:
We disconnected the 4 ESC and it does not turned on (see attached files). 
What does it mean?

[Archive.zip](https://github.com/intel-aero/meta-intel-aero/files/2183646/Archive.zip)

talma, 2018-07-17:
Any comment on the above ?

lucasdemarchi, 2018-07-17:
From the video you sent I see the board is turned on. Thus it turn off when you release the button? If so the cable that connects the IO board (the one on the top) to the PDB (where you are pressing the on button) may be loose or broken. Try reconnecting it. Another cable that may be loose or broken is the flat cable going from the IO board to the Aero board. Try reconnecting it too (careful not to break it).