intel-aero/meta-intel-aero #367: lidar lite slow read rate

ambikaverma, 2018-07-17:
Hi
I'm trying to connect lidar lite v3 to the COMPASS port - have tried coneecting through an i2c splitter as well as directly to the compass port with mag0 removed (doesn't seem to be a i2c splitter issue).

I do see the distance sensor in QGC MavLink inspector and see the data change, but the rate of change is extremely slow because of which the drone for example held at 1m the altitude measured is still 0.4m resulting in EKF HGT error.

In mavlink inspector it shows that the data is read at 10Hz but when actually seeing the value, it changes very slowly and gets stuck at the same value for 3-5 seconds.

I tried connecting the lidar lite to Arduino Uno again using I2C but here I can clearly see the data updating at around 400Hz (the default update rate of lidar lite v3).

Is the 10Hz sampling rate at compass port a fixed value? Should I be using telemetry port to get a higher update rate?
Aero version -
bios - 1.6.2
Flight controller - 1.8

zehortigoza, 2018-07-17:
The driver reads the lidar at 20hz but MAVLink stream is limited to 10hz(you can check and change both doing changes in PX4).

Did you added the capacitor to VCC and GND as lidar datasheet recomends?
It improves a little bit but Aero do not play well with Garmin lidar lite(https://docs.px4.io/en/flight_controller/intel_aero.html#lidar_lite) it is recommended to switch to LeddarOne instructions here: https://docs.px4.io/en/flight_controller/intel_aero.html#leddarone

zehortigoza, 2018-07-17:
There is a 2 new I2C transactions(LL40LS_SIGNAL_STRENGTH_REG and LL40LS_PEAK_STRENGTH_REG) https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/ll40ls/LidarLiteI2C.cpp . These new reads could be failing causing no orb publish, that is why it gets stuck for a long time. You can check that by running perf command into nuttx terminal, you should have a lot of ll40ls_i2c_comms_errors.

The peaks is I2C communication problems were some bits gets flipped, it can happen when reading or writing from slave.

zehortigoza, 2018-07-18:
Huum by the perf output it is not I2C communication problem, I don't have a lidar lite to debug anymore.
If you want you can debug it your self, bringing the serial console would help you to that(https://github.com/intel-aero/meta-intel-aero/issues/321) or you could go back and test with the PX4 firmware that ships with Aero images.

ambikaverma, 2018-07-26:
update -
went back to ver 1.8.0 and now configured the telemetry port as i2c using the jam file provided here -
https://www.instructables.com/id/Intel-Aero-Drone-Altitude-and-Position-Hold-Using-/

and the lidar lite now works at the 10Hz rate. 
not entirely sure why it didn't work when connected directly to compass port. Compass port configured differently? will check the jam files to see any differences between compass i2c and telemetry i2c.
Thanks for the help.
Closing the issue.

closed 2018-07-26 by ambikaverma