intel-aero/meta-intel-aero #368: here+ rtk not detected

ambikaverma, 2018-07-21:
I'm trying to connect here+ rtk to intel aero.
currently working with px4 ver 1.6.5 shipped with aero images.

the rover is running ver 1.3 and is detected in ubox u-center software.
OGC recognizes here+ base module but the rover module is never recognized.

I try checking gps status in nuttx shell and consistently get GPS status NOT OK.
Any guidance?
magnetometer seems to be detected and i get a large offset error, since I didnt orient it properly, but in perf command output I don't see hmc5833 mag. I only see MPU9250.

I'm yet to try with px4 ver 1.8.0 and ardupilot.

zehortigoza, 2018-07-21:
Maybe it needs more time to fetch the almanac.
Other user had no problems with the GPS part, he only had problems with one of the compass, check here: https://github.com/intel-aero/meta-intel-aero/issues/248

ambikaverma, 2018-07-26:
fixed the issue now.
I tried all the firwares px4 1.6.5, 1.8.0 and even ardupilot but it didn't work with any of them.
Also ensured that the GPS was set to use UBX protocol and tried setting a few different baud rates, but nothing worked.
In the end switched the RX and TX connection and it finally started working. 
the pins are switched on my gps unit and incorrectly marked.
thanks for the help! closing the issue now.

closed 2018-07-26 by ambikaverma