intel-aero/meta-intel-aero #370: how to access GPS or waypoint data directly in intel aero?

ambikaverma, 2018-07-31:
Hi

I want to start running a python script only when the uav is close to a waypoint (which I can track by comparing the target gps location with current gps data).

Any recommendations for what I can use to access this data directly on intel aero (while still running QGC and not using ubuntu on itel aero)?

Thanks!

lucasdemarchi, 2018-07-31:
I don't fully understand what you want to do, but it seems like you want to have a script running and comparing the GPS positions. When it reaches a certain mission waypoint you trigger an action that's basically firing your dedicated script.

There's a mavlink message from the flight stack notifying you when it considers the waypoint "done". You can use that instead of comparing the GPS coordinates, too.

ambikaverma, 2018-07-31:
Yup that is exactly what I want to do. Sorry if I wasn't clear.

Could you tell me what mavlink message is it?
And to monitor the message which port should I be reading from/ listening to?

lucasdemarchi, 2018-07-31:
yes. that one.... See the example on mavlink-router to use TCP port 5760 and make sure to start your service on boot.

ambikaverma, 2018-08-10:
Thanks I was able to test the tcp example - https://github.com/intel/mavlink-router/blob/master/examples/heartbeat-print-tcp.py

Could you point out why I need to start this on boot? 
if i start this tcp script afterwards it still gives me the current data right (from what I saw)?
also, to start this on boot I will have to create a systemd service right? similar to the aero-optical-flow service?

ajeyaajeya, 2019-05-18:
I have the same question as the previous one.