intel-aero/meta-intel-aero #371: how to capture depth data from R200?

ambikaverma, 2018-07-31:
Hi

I tried running cpp-headless to capture depth data from R200 on intel aero as per -
https://github.com/intel-aero/meta-intel-aero/wiki/06-Cameras-and-Video
but the command is not recognized. 
I just ssh to intel-aero.local and run the above command from /home/root.

My version is -
[image]

Any other methods to capture depth data?

damadamn, 2018-08-02:
I've got the same problem about `cpp-headless`, but there is for example `cpp-capture` which provides some different view of your RealSense, including the depth data

ambikaverma, 2018-08-02:
Is cpp-capture available on intel aero? I'll try and run it anyway.
if not where can I find it?

Thanks!

ambikaverma, 2018-08-04:
for cpp-capture I'm getting following error -

[image]

also tried -
sudo gst-launch-1.0 v4l2src device=/dev/video11 num-buffers=1 ! jpegenc ! filesink location="rs.jpg"
identify rs.jpg 

but all I get is a green frame.

ambikaverma, 2018-08-04:
Another issue that I'm facing -
[image]

the same command worked fine earlier, but now I get this error.
got this error after trying to run cpp-capture as discussed here https://github.com/intel-aero/meta-intel-aero/issues/371 (which itself also failed).

and in QGC I can only see data from rsdepth stream. Nothing from video13, rsir and rsir2 streams.

Did running cpp-capture change any settings (even though it failed)?