intel-aero/meta-intel-aero #372: Optical Flow active but not working

rr2-ip, 2018-08-02:
Hi,
I've installed a LeddarOne rangefinder on my Intel RTF drone and I am having trouble getting the optical flow to work.

I can see the raw distance measurements under the DISTANCE_SENSOR topic in the MAVLink Inspector of QGroundControl. 
When optical flow is active and I check the status, I see messages as shown in the attached picture. 
[image]

Does anyone have experience in resolving this issue?

FYI:` aero-get-version.py` returns:
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Ubuntu 16.04.5 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.8.0

zehortigoza, 2018-08-02:
The "Number of good matches: x, desired: y" means that the floor don't have enough texture to optical flow to work, this also happens in a floor with good texture but the drone is still too close to the ground. Try to hand-held it at 0.5m and check if this messages continue to be printed, best way to check the messages is stop the aero-optical-flow service and start it manually.

About the "FPS below minimum, actual=X minimum=70" if this happens just in the first seconds it is okay otherwise it is a problem, I guess @broody had found this problem too maybe he can help.

zehortigoza, 2018-08-03:
Run by hand ./aero-optical-flow this way you can see all the messages not only the last ones.

Check the optical-flow quality as explained here https://github.com/intel-aero/aero-optical-flow/blob/master/README.md but I guess you floor don't have enough texture to run optical-flow with aero camera.

rr2-ip, 2018-08-03:
Quality is around 180-190. Here is my floor - Would you expect it to work on this floor?

[image]

When running a continuous optical flow stream in the terminal, I see the image in my first post being repeated. When lifted off the ground, I see a continuous stream of FPS below minimum (as in the image in my second post).

rr2-ip, 2018-08-03:
I might also add that when the drone is moved somewhat rapidly, I get:
[Debug] Camera timeout, restarting...

zehortigoza, 2018-08-03:
This is also normal: [Debug] Camera timeout, restarting...

If it stays consistently with this quality around 180-190 it should work.
Just to make sure, confirm that you set 2 parameters: EKF2_AID_MASK and EKF2_HGT_MODE.

What happens after you set the position mode?

rr2-ip, 2018-08-03:
Yes the quality is consistent.
I have set the following parameters:
EKF2_AID_MASK = 2
EKF2_HGT_MODE = Range Sensor (We're not using GPS - GPS antenna is disconnected)

Do you mean setting position mode on the remote controller?

rr2-ip, 2018-08-03:
When I put the drone in position mode I get an error in QGC:
REJECT POSITION CONTROL

zehortigoza, 2018-08-03:
Everything looks okay, I would try to fly in altitude to verify leddar one readings, if it still works I suggest you to add debug messages in PX4 code to get the reason why it is being rejected, maybe this changes would be accept upstream if done right.

rr2-ip, 2018-08-03:
Ok I'll give that a go. How do I add debug messages to the PX4 code?

rr2-ip, 2018-08-08:
Hi @zehortigoza ,
I've been doing some test flying.
Distance sensor is working and visible in QGC.
Still receiving REJECT POSITION CONTROL error upon entering position flight mode.
Can't take off in altitude mode even at full throttle - motors still arm and there are no error messages in this mode whatsoever.

gzmolvr, 2018-08-09:
I have the same exact issue. Distance shows correct values in QGC. Always get REJECT POSITION CONTROL. No take off in height mode. Manual mode works fine.

zehortigoza, 2018-08-09:
Looks like there is some regression in PX4, I can't check this in the next few weeks. Maybe open a issue in PX4, some other PX4 developer could check that.
Or try with older releases, I had it working really well(https://www.youtube.com/watch?v=sMrkOJoF-jM) in PX4 1.6~1.7.

rr2-ip, 2018-08-12:
All other problems are secondary to the fact that I receive `[Error] FPS below minimum` when running `aero-optical-flow`. This error is present on PX4 v1.6.5, 1.7.3 and 1.8.0, even after adding the LeddarOne driver (which does not show up in QGC on versions prior to 1.8.0).

rr2-ip, 2018-08-13:
I get the error around 10 times per second. The quality appears to be consistently above 100. The drone is still rejecting position control.

geetesh, 2018-08-24:
I have the same issue i.e. reject position, I am using benewake tfmini, on latest px4 build as of Aug 23rd 2018. I know tfmini has issues at less than 30cm altitude so I climb upto a meter, after some settling time, then switch to altitude hold. Which is okayish, it oscillates up and down a bit. Then position mode is always rejected 😬