intel-aero/meta-intel-aero #376: optical flow position mode - "no local position" error

ambikaverma, 2018-08-05:
Hi

Im trying to test optical flow on intel aero. My setup is as follows -

px4 - ver 1.8.0
bios - latest yocto image.
indoors, when the uav is stationary.

I start optical flow manually using ./aero-optical-flow command to see if the features/texture is good enough. Looks like its good enough since I don't get any desired number of feature prompts. Also the quality is on 180-200 range.

And on arming the flight mode does change to position hold.
But, instantly I get "no local position" error. I can share screenshots of above if required.

Any suggestions on the cause for this?
Thanks!

ambikaverma, 2018-08-07:
any suggestions for above issue? Im still getting no local position error.
 ekf_aid_mask = 2
ekf_hgt_mode = range sensor

optical flow quality is good 180-200 and I'm able to switch to position hold mode, but instantly goes into failsafe for "no local position"

zehortigoza, 2018-08-07:
Looks very similar to https://github.com/intel-aero/meta-intel-aero/issues/372
as said in that issue, try to switch to altitude mode to make sure everything lidar is working other than that you will have to debug PX4.

AlexisTM, 2018-08-23:
Do you have a laser connected?