intel-aero/meta-intel-aero #381: QGC can receive the video but cannot connect with the drone

ArlenCHEN, 2018-09-25:
I met a problem. Every time I run QGC, it will automatically detect the video stream from intel aero, but QGC cannot connect with the drone. It is always showing that "waiting for vehicle connection". But I can see the video stream on the left-bottom corner.

Another thing is once the QGC receive the video stream, it seems that I cannot use the camera by other process. For example, I cannot start up the ORB-SLAM2 on intel aero once I run QGC on my laptop. But if I shut off the QGC, camera will be detected and I can run ORB-SLAM2 on the drone!

How to connect QGC with drone while I can receive the video stream?
Hope some guys help me!

zehortigoza, 2018-11-06:
Check if mavlink-router is running and if FPGA is flashed.

About the camera only being access by one process that is the expected behavior of the driver.

lucasdemarchi, 2018-11-06:
> I met a problem. Every time I run QGC, it will automatically detect the video stream from intel aero, but QGC cannot connect with the drone. It is always showing that "waiting for vehicle connection". But I can see the video stream on the left-bottom corner.

In other words: QGC can connect to the Aero Compute Board, but it can't connect to the  Flight Controller.  As @zehortigoza said, confirm the FPGA is flashed correctly.

> Another thing is once the QGC receive the video stream, it seems that I cannot use the camera by other process. For example, I cannot start up the ORB-SLAM2 on intel aero once I run QGC on my laptop. But if I shut off the QGC, camera will be detected and I can run ORB-SLAM2 on the drone!

You can't use them simultaneously as is. Some options:

1) Either stream or use orb-slam2
2) Create a fake device that will have the copied data. See for example https://www.timdejong.nl/blog/use-webcam-two-applications-under-linux-simultaneously-using-v4l2loopback
3) Change camera-streamer to relay the data to a socket/shm and use that as your source in orb-slam2

I think (2) is the easiest one, but your mileage may vary.