intel-aero/meta-intel-aero #382: Gazebo simulation

Ymonse, 2018-10-05:
Hello,

Can we use the gazebo simulation without the intel aero ready to fly UAV ? 

I tried to clone that package : git clone git@github.intel.com:lecampos/aero_simulator.git, but I get "ssh: could not resolve hostname github.intel.com".

I'm on a ubuntu 16.04 virtual machine.

Thank you.

lucasdemarchi, 2018-10-05:
@guermonprez wrong link to the repository? You may need to migrate it under github.com/intel-aero and update the docs.

gbarsih, 2018-10-08:
I'm having the same problem, could we have a temporary link to the simulator while you update the location?

Ymonse, 2018-10-10:
Yes it would be great if you could provide a temporary link, because I still have the same problem.

stribuda, 2018-10-16:
So what is the solution please?