intel-aero/meta-intel-aero #386: unable to save video

mymultiverse, 2018-10-10:
I am running following command but it is just showing warning. 

 

```
sudo gst-launch-1.0 -e v4l2src device=/dev/video13 num-buffers=2000 ! autovideoconvert format=i420 width=1920 height=1080 framerate=30/1 ! vaapih264enc rate-control=cbr tune=high-compression ! qtmux ! filesink location=encoded_video.mp4
WARNING: erroneous pipeline: no element "autovideoconvert"

```

hooverryan, 2019-02-12:
I got mine to work using the following command:
`sudo gst-launch-1.0 -e rtsp location=rtsp://192.168.8.1:8554/video13 ! rtph264depay ! h264parse ! mp4mux ! filesink location=video.mp4`

mymultiverse, 2019-02-25:
> I got mine to work using the following command:
> `sudo gst-launch-1.0 -e rtsp location=rtsp://192.168.8.1:8554/video13 ! rtph264depay ! h264parse ! mp4mux ! filesink location=video.mp4`

In this case, I am getting 

`WARNING: erroneous pipeline: no element "rtsp"`