intel-aero/meta-intel-aero #393: Consolidated install instructions for PX4 on compute board only

Seanmatthews, 2018-11-05:
Is there a set of instructions for installing PX4 on the compute board only? I found FPGA flashing instructions for compute board only [here](https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup#flashing-the-fpga), but do those allow for PX4 to run necessary components on the FPGA? Any information you can provide will be helpful. Thanks!

zehortigoza, 2018-11-06:
You want to run PX4 in the Atom processor? There is no PX4 port to run into Aero compute board, there is a partial port to run Ardupilot but not fully tested and supported.

lucasdemarchi, 2018-11-06:
The sensors on the compute board are mostly intended for other functions, not for the flight stack.  As @zehortigoza said, there's a port of ArduPilot that runs on it and make use of the sensors (on ArduPilot you should pass the board as `./waf configure --board=aero` to build for the).  It works, but since there's very limited amount of UARTS (1), you will have to choose between having RC or GPS. Or add another one through usb-serial.

In this post here https://discuss.ardupilot.org/t/using-sololink-for-linux-based-boards/21719 I used a wifi-based RC to workaround that issue. And since I'm controlling the Aero RTF, the connection to the ESCs is also UART-based, so I had to use a usb-serial for the GPS nonetheless.

Seanmatthews, 2018-11-12:
@zehortigoza Moreso, I was wondering if there was a port that makes use of the on-board FPGA to run PX4.

zehortigoza, 2018-11-12:
@Seanmatthews You mean synthesize microcontroller in FPGA and run PX4 there?! The FPGA in Aero don't have enough logic blocks to synthesize something that can run PX4.

Seanmatthews, 2018-11-12:
@zehortigoza What's the FPGA for the Aero compute board used for? How many logic blocks do you suppose would be needed to emulate the STM32 microcontroller that the RTF drone uses?

zehortigoza, 2018-11-12:
In the AeroRTF, FPGA is used to route some signals from the external connector to Atom also it works as battery voltage sensor and there still more than 50% of logic blocks available to user use.

I don't know how many would be need or even if the STM32 design is available somewhere to be synthesized in FPGAs maybe you can find a Cortex-M3 but not a STM32. And even if you find it, you would need to connect a IMU in the external connector... so not practical at all