intel-aero/meta-intel-aero #395: After installing Ubuntu 16.04.3 LTS keyboard and mouse are not detected

RohanKurane, 2018-12-26:
@zehortigoza @lucasdemarchi 

I first upgraded Yochto to the latest kernel using the iso file on intel website. Then flahsed the BIOS, FPGA and FC. 
Then I installed Ubuntu 16.04.3 LTS which went fine. 

After completing the installation, when asked to restart , upon restart the keyboard and mouse connected to the OTG cable are not detected anymore so there is no way for me to interact with the RTF drone. 

Several other people have asked the same question before but I do not have control of the keyboard or mouse anymore. Only when I power up the drone do I have control of them. 

What can I do here ?

RohanKurane, 2018-12-27:
One more question - at this moment with ubuntu installed but keyboard and mouse not detected, I have not yet added the Intel Aero repository as documented in the wiki steps -
https://github.com/intel-aero/meta-intel-aero/wiki/90-%28References%29-OS-user-Installation#intel-realsense-sdk---legacy-for-r200

At this moment I do not see the AERxxxxx wifi hotspot. Would it show after adding the Intel Aero repository ? Or with Ubuntu will it not show now and I need to use other Wifi networks ?

lucasdemarchi, 2018-12-27:
Easier way is to reflash again and add the repository mentioned. This will install the patched kernel to make USB otg to switch to host mode

RohanKurane, 2018-12-27:
@lucasdemarchi 
not sure I understood. 
Reflash Yocto ? Even if I do, I cant run 'apt-get' commands on Yocto.  
And would that help me gain access to the keyboard/mouse after installing ubuntu ?

RohanKurane, 2018-12-27:
If you are referring to reflash ubuntu 16.04.3, I did try a re-install of it. But still cant access the mouse/keyboard after the install. 
The drone is not connected to any Wifi upon installation complete so I cannot ssh to it either from my local laptop. Nor can I see the AERO-xxxx hotspot in the list of wifi connections on my local laptop. 
@lucasdemarchi

RohanKurane, 2018-12-27:
Here is a screen shot from my BIOS for USB configuration. Does this look okay to you ? 
[image]

RohanKurane, 2018-12-27:
After a restart is done, this screen comes up and right before this screen comes up, the keyboard green light goes off. So somewhere the USB OTG is disabled. Maybe a kernel incompatibility or something.  So I cant really interact with the Intel aero after the restart. 
[image]

RohanKurane, 2018-12-27:
The funny thing is, when I started to install Ubuntu 16.04.3 LTS today the keyboard and mouse were detected just fine. I was able to get to the BIOS menu, boot from usb stick and go thru the usual ubuntu install. 
After the install completed, it asked me to restart which I believe is normal. I was able to use the mouse to click the Restart Now button. 
After the restart, it asked me to remove the USB media and hit Enter (which I believe is also normal for installation via USB or CD/DVD media). I did that and hit enter on my keyboard. 
So the keyboard and mouse were detectable till then. 
Then the screen went blank for 2 secs and came up with the login screen to enter my password. 
When the screen went blank, thats when the keyboard and mouse stopped working  I would imagine. 

So somehow the USB OTG connection was disabled during that time and hence the keyboard and mouse connected to it were not responsive anymore. Any idea what may have happened ? I suspect some kind of ubuntu BIOS config setting for Intel Aero that I am unaware of. Or perhaps kernel incompatibility. But if you can and others may have installed the same ubuntu version on Intel Aero (after upgrading it to the latest FPGA, FC and BIOS), then why am I hitting this issue ?