intel-aero/meta-intel-aero #396: How to use apt-get install on intel-aero ?

RohanKurane, 2018-12-30:
@lucasdemarchi @zehortigoza 

I have installed ubuntu 16.04 LTS on intel-aero. 

Can you tell me how I can use "apt-get install" to download any packages on intel-aero ? It requires internet connection I believe and I am connected to the Aero hotspot on the intel-aero. 
What can I do to use "apt-get install" on the intel-aero ?

Thanks 
Rohan

lucasdemarchi, 2018-12-30:
You will need to configure  it to connect to an existent network instead of using it as an access point

RohanKurane, 2018-12-30:
How can i do that? 
When I start it up and try the ubuntu network manager all i see is the hotspot.

RohanKurane, 2018-12-31:
FYI.. here is how it needs to be done -
https://github.com/intel-aero/meta-intel-aero/wiki/08-Aero-Network-and-System-Administration#networking-internet-access