intel-aero/meta-intel-aero #397: intel-aero kernel serial8250: too much work for irq39

RohanKurane, 2019-01-02:
@lucasdemarchi  @zehortigoza 
I noticed several folks including you have hit this issue. 

In my case I am not doing anything with mavlink. I am running my application on the intel aero and see this message which I believe is causing failures in my application. 

I have updated the intel aero to the latest all and Ubuntu 16.04 LTS. 

What can I do here to avoid the message ?

lucasdemarchi, 2019-01-02:
If you are not using mavlink, do you have anything using the uart? Is mavlink-router running?

RohanKurane, 2019-01-02:
Here is the output from cat /proc/interrupts

 39:      14356     167389          0          0   IO-APIC   39-fasteoi   serial

Clearly there are a tons of interrupts going on for irq39. 

What can I do here ? Can I clear those numbers for CPU0 and CPU1 ? Or do anything else ?

RohanKurane, 2019-01-02:
@lucasdemarchi
mavlink-router is running

rohan@intel-aero:~$ systemctl status mavlink-router
● mavlink-router.service - MAVLink Router
   Loaded: loaded (/lib/systemd/system/mavlink-router.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-12-31 02:32:53 PST; 10min ago
  Process: 725 ExecStartPre=/bin/bash -c echo 4 > /sys/class/tty/ttyS1/rx_trig_bytes (code=exited, status=0/SUCCESS)
 Main PID: 806 (mavlink-routerd)
   CGroup: /system.slice/mavlink-router.service
           └─806 /usr/bin/mavlink-routerd

Dec 31 02:32:53 intel-aero systemd[1]: Started MAVLink Router.
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Open UDP [4] 192.168.8.85:14550
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Open UDP [5] 192.168.7.255:14550
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Open UDP [6] 0.0.0.0:80550 *
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Open UART [7] /dev/ttyS1 *
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: UART [7] speed = 921600
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Open TCP [9] 0.0.0.0:5760 *
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Baudrate 921600 responded, keeping it
Dec 31 02:32:53 intel-aero mavlink-routerd[806]: Logging target system_id=1 on 00023-2018-12-31_02-32-53.ulg
Dec 31 02:32:58 intel-aero mavlink-routerd[806]: 4 messages to unknown endpoints in the last 5 seconds

RohanKurane, 2019-01-02:
I stopped the mavlink-router and it did not help.

RohanKurane, 2019-01-03:
I was able to get around the problem by changing some code in my application. 
But I do see these messages from time to time. 
Is there any way to mitigate this ?

closed 2019-01-03 by RohanKurane